MichaIng/DietPi #972: Orange pi lite boot can not automatically connect wifi

syfxlin, 2017-05-21:
Orange lite restart is always not connected wifi, through the phone usb network sharing orange lite with ifconfig can not see usb0 information, but with lsusb can see the phone has been connected to the orange lite. Please help me. Thanks.

syfxlin, 2017-05-21:
Armbain does not exist these problems

Fourdee, 2017-05-21:
@syfxlin 

> Armbain does not exist these problems

Interesting, we use ARMbian to build the image. Most likely ARMbian image is running a different kernel that "compliments" WiFi.

Kernel version installed on DietPi?
```
uname -a
```

Kernel version installed on ARMbian?
```
uname -a
```

syfxlin, 2017-05-22:
@Fourdee Kernel version is the same, now pi can not be recognized as a network interface usb port. Is there any way to connect to pi through usb?

ThomasKaiser, 2017-06-29:
@syfxlin 

Easy, remove `wlan0` definition below `/etc/network/interfaces`, then

    sudo apt --no-install-recommends install network-manager 
    sudo nmtui-connect

> now pi can not be recognized as a network interface usb port. 

If you're talking about connecting your OPi Lite with a Micro USB cable from a PC and get a serial console (`g_ether` mode) then this is just another Armbian feature most probably not available here.

closed 2018-02-22 by Fourdee