MichaIng/DietPi #973: veracrypt ? is possible add tho dietpi?

Ox16, 2017-05-21:
Is possible add veracrypt

Fourdee, 2017-05-22:
> https://www.veracrypt.fr/en/Home.html

@Ox16 

Thanks for request 👍 We'll take a look when possible.

DJCrashdummy, 2017-05-29:
well i'm not sure if it is appropriate to mention it here, but what about [zuluCrypt](https://mhogomchungu.github.io/zuluCrypt/)?

> zuluCrypt does hard drives encryption and it can manage PLAIN dm-crypt volumes, LUKS encrypted volumes, TrueCrypt encrypted volumes and VeraCrypt encrypted volumes.

it has a CLI and a GUI, is translated into several languages and since debian stretch (9.0) it is also available via the repositories.

so i guess this issue will be solved at the latest by porting to debian stretch without any additional work, extra compiling, etc.

WolfganP, 2017-05-29:
My 2 cents on the discussion. Veracrypt is multiplatform, meaning you can move the virtual drives between platforms when needed, and it was properly audited for security last year (http://blog.quarkslab.com/security-assessment-of-veracrypt-fixes-and-evolutions-from-truecrypt.html), plus there's experience around on installing it on a raspberry (ie https://randomgooby.wordpress.com/2016/09/10/quick-tip-veracrypt-on-a-raspberry-pi-2-update-veracrypt-1-18/) to avoid starting from scratch.

DJCrashdummy, 2017-05-29:
@WolfganP i think you missed the point: _(a software and a format are two different things!)_

- zuluCrypt is about working with VeraCrypt-drives and even others, so it is more flexible than the VeraCrypt-software itself. - ...on other OSes you can/must stay with the VeraCrypt-software to read VeraCrypt-drives.
- when providing for and installing on debian stretch there is no (special) experience needed at all! - a simple `apt install zulucrypt-cli` and/or `apt install zulucrypt-gui` like for all other software you want to install is enough.
- yes you are right, Truecrypt/VeraCrypt was audited, but there are never ending discussions how much sense such audits make... BUT i don't want to start such a discussion here, not least because i can't make a clear statement about this topic because i'm neither a developer nor a security researcher.
_for example and explanation: at linux-environments very few care about Truecrypt/VeraCrypt because THE way to go for encrypting drives is the omnipresent LUKS, which was never "official audited", but no one would call it insecure..._

oh! i nearly forgot to mention - maybe it makes you feel better using zuluCrypt - that [zuluplay](https://github.com/mhogomchungu/zuluplay) (the backend for handling VeraCrypt-drives) is forked from [tcplay](https://github.com/mhogomchungu/tc-play) which was also partly developed by the VeraCrypt-developers.

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

hfavisado, 2021-01-16:
Hi! It'd be great if we could have this available in dietpi-software. I'm using an intel nuc to share a couple of HDDs that I used previously as USB disks in windows, so encryption was handled by veracrypt.