MichaIng/DietPi #974: DietPi-Software | Upgrade NextCloud 12

Fourdee, 2017-05-23:
> https://github.com/nextcloud/server/releases/tag/v12.0.0

Fourdee, 2017-05-23:
- 🈯️ Test install fine, other than version link update, no code changes needed.

closed 2017-05-23 by Fourdee

molnarti, 2017-06-01:
does the uninstall/reinstall procedure maintain the nextcloud settings? like user, data location, quota, apps, etc. or a complete setup from scratch is needed?

Fourdee, 2017-06-02:
> does the uninstall/reinstall procedure maintain the nextcloud settings? like user, data location, quota, apps, etc.

Nope

>  or a complete setup from scratch is needed?

Unless NextCloud UI has a feature to update, a fresh reinstall is required.

molnarti, 2017-06-05:
> Unless NextCloud UI has a feature to update, a fresh reinstall is required.
actually has and i used it in the past for minor updated (11.0.2 > 11.0.3), but for 12 it does not offer the upgrade

molnarti, 2017-06-09:
tried updating via the uninstall reinstall path. this is what I am getting:

`[Info] Creating MySql DB for nextcloud`
` [Ok] Create database nextcloud | Completed`
`Nextcloud is not installed - only a limited number of commands are available`

then system gets stuck without any response