MichaIng/DietPi #975: RPi - USB drive removal and reinsert doesn't mount device properly

danriches, 2017-05-23:
I've tried numerous fstab entries and with and without samba installed to no avail. The configuration works fine with Raspbian Jessie though, albeit Raspbian is a nightmare to get to boot in readonly mode hence why I'm trying out DietPi. I've attached the output from journal regarding the USB flash drive mount problem, this is without writing to the drive only reading at the moment. Any suggestions would be very much welcome, Thanks, Dan...

Of particular note is the mismatch of device identifiers (sdb and sda) in the following lines near the end of the log:
Apr 21 16:14:27 DietPi kernel: sd 1:0:0:0: [sdb] Attached SCSI removable disk
Apr 21 16:14:44 DietPi kernel: FAT-fs (sda1): Directory bread(block 30600) failed

JOURNAL OUTPUT
root@DietPi:~# apt-get remove usbmount
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  lockfile-progs pmount
Use 'apt-get autoremove' to remove them.
The following packages will be REMOVED:
  usbmount
0 upgraded, 0 newly installed, 1 to remove and 4 not upgraded.
After this operation, 119 kB disk space will be freed.
Do you want to continue? [Y/n] y
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "0.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
(Reading database ... 31168 files and directories currently installed.)
Removing usbmount (0.0.22) ...
root@DietPi:~# reboot
login as: root
root@192.168.8.184's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ───────────────────────────────────────
 DietPi     | 16:09 | Fri 21/04/17
 ───────────────────────────────────────
 V150       | RPi 3 Model B (armv7l)
 ───────────────────────────────────────
 IP Address | 192.168.8.184
 ───────────────────────────────────────

 Created by : Daniel Knight
 Web        : http://DietPi.com
 Twitter    : http://twitter.com/dietpi_
 Donate     : http://goo.gl/pzISt9
 DietPi's web hosting is powered by: MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

root@DietPi:~# ls /mnt/DISK
HDClinicalPacePort  System Volume Information
root@DietPi:~# ls /mnt/DISK
root@DietPi:~# ls /mnt/DISK
root@DietPi:~# ls /dev
autofs           gpiochip1     loop7               ram1    raw      tty13  tty28  tty42  tty57      vc-cma  vcsa5
block            gpiochip2     mapper              ram10   sdb      tty14  tty29  tty43  tty58      vc-mem  vcsa6
bsg              gpiomem       mem                 ram11   sdb1     tty15  tty3   tty44  tty59      vchiq   vhci
btrfs-control    hwrng         memory_bandwidth    ram12   serial1  tty16  tty30  tty45  tty6       vcio    watchdog
[... 411 lines cut ...]
Apr 21 16:14:27 DietPi kernel:  sdb: sdb1
Apr 21 16:14:27 DietPi kernel: sd 1:0:0:0: [sdb] Attached SCSI removable disk
Apr 21 16:14:44 DietPi kernel: FAT-fs (sda1): Directory bread(block 30600) failed
Apr 21 16:14:44 DietPi kernel: FAT-fs (sda1): Directory bread(block 30601) failed
Apr 21 16:14:44 DietPi kernel: FAT-fs (sda1): Directory bread(block 30602) failed
Apr 21 16:14:44 DietPi kernel: FAT-fs (sda1): Directory bread(block 30603) failed
Apr 21 16:14:44 DietPi kernel: FAT-fs (sda1): Directory bread(block 30604) failed
Apr 21 16:14:44 DietPi kernel: FAT-fs (sda1): Directory bread(block 30605) failed
Apr 21 16:14:44 DietPi kernel: FAT-fs (sda1): Directory bread(block 30606) failed
Apr 21 16:14:44 DietPi kernel: FAT-fs (sda1): Directory bread(block 30607) failed
Apr 21 16:14:44 DietPi kernel: FAT-fs (sda1): Directory bread(block 30608) failed
Apr 21 16:14:44 DietPi kernel: FAT-fs (sda1): Directory bread(block 30609) failed
Apr 21 16:14:58 DietPi smbd[2227]: pam_unix(samba:session): session closed for user nobody
Apr 21 16:15:01 DietPi kernel: fat__get_entry: 6 callbacks suppressed
Apr 21 16:15:01 DietPi kernel: FAT-fs (sda1): Directory bread(block 30600) failed
Apr 21 16:15:01 DietPi kernel: FAT-fs (sda1): Directory bread(block 30601) failed
Apr 21 16:15:01 DietPi kernel: FAT-fs (sda1): Directory bread(block 30602) failed
Apr 21 16:15:01 DietPi kernel: FAT-fs (sda1): Directory bread(block 30603) failed
Apr 21 16:15:01 DietPi kernel: FAT-fs (sda1): Directory bread(block 30604) failed
Apr 21 16:15:01 DietPi kernel: FAT-fs (sda1): Directory bread(block 30605) failed
Apr 21 16:15:01 DietPi kernel: FAT-fs (sda1): Directory bread(block 30606) failed
Apr 21 16:15:01 DietPi kernel: FAT-fs (sda1): Directory bread(block 30607) failed
Apr 21 16:15:01 DietPi kernel: FAT-fs (sda1): Directory bread(block 30608) failed
Apr 21 16:15:01 DietPi kernel: FAT-fs (sda1): Directory bread(block 30609) failed
Apr 21 16:15:14 DietPi smbd[2232]: pam_unix(samba:session): session closed for user nobody

Fourdee, 2017-05-23:
@danriches 

Hi,

We have tool (```dietpi-drive_manager```) which will allow you easily add USB drives, also applies auto mount (```x-systemd.automount```) to the fstab entry, removing the need for ```usbmount```:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=673#p2898

Try clearing your ```/etc/fstab``` changes, then run ```dietpi-drive_manager``` to setup the drive.

> Directory bread(block 30605) failed

Possible issue with filesystem. Try repairing the drive with:
```
dosfsck -w -r -l -a -v -t /dev/sda1
```

danriches, 2017-05-24:
There is a problem with using the drive manager as I can't mount by name only by uuid, I need this to be a name such as DISK and shared out on samba. Otherwise all my products will need the usb devices serial number set on format and not just a nice quick and easy rename. As I've said, this works like a charm on Raspbian Jessie and the only extra thing I need is to make the rootfs readonly which has given me nothing but trouble on Jessie.

The filesystem check ran up errors even though this is a brand new flash drive formatted as FAT32 as that's what an external piece of equipment needs, I cannot use NTFS or ext4 etc. If I format this on a Windows box and use in Jessie it works absolutely fine, no errors and the automount by name works.

It would seem to me that an essential package or configuration is missing for this all to work and the fact that you can't seem to mount by name in drive manager is a real issue for me. Personally, I'm going to look at other distros to see what they offer before probably biting the bullet and going back to Jessie and the awful mess of getting a readonly rootfs working.

Thanks for the response and if there's a resolution to my issue then please let me know, I'd much rather use DietPi for it's boot time and ease of use. You've done a cracking job :)

Dan

Fourdee, 2018-02-22:
Outdated.

closed 2018-02-22 by Fourdee