MichaIng/DietPi #976: Documentation | Write to SD card using macOS

msdos, 2017-05-25:
In these instructions, there's no suggestion for macOS

http://dietpi.com/phpbb/viewtopic.php?f=8&t=9#p9

There's a small application with a GUI interface that can do it:

https://www.tweaking4all.com/software/macosx-software/macosx-apple-pi-baker/

[image]

I've only tested in RPi, but they say it can work in other devices as well. It's really simple: I just select my SD Card, the IMG file and click "Restore Backup": done!

sootstone, 2017-05-25:
Nice idea :+1: 

While on this subject, I see quite a few distros recommending [Etcher](https://etcher.io/) now (even raspbian), which could be a nice way to standardize the installation across Windows, Linux and OSX.

msdos, 2017-05-25:
This seems really nice @sootstone, actually I think yours is a better idea 😆

Fourdee, 2017-05-27:
@msdos 

> This seems really nice @sootstone, actually I think yours is a better idea 

Agree 👍 

@sootstone 

I've updated our online doc for Etcher, seemed the best choice for all OS's
http://dietpi.com/phpbb/viewtopic.php?f=8&t=9&p=9#p9

I'll mark this as closed.

closed 2017-05-27 by Fourdee