MichaIng/DietPi #977: DietPi doesn't boot on Orange Pi Plus 2

it-praktyk, 2017-05-25:
An image downloaded today doesn't boot on Orange Pi Plus 2

Boot log from serial console

```
U-Boot SPL 2017.01-armbian (Feb 02 2017 - 03:04:04)
DRAM: 2048 MiB
Trying to boot from MMC1

U-Boot 2017.01-armbian (Feb 02 2017 - 03:04:04 +0100) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Xunlong Orange Pi Plus / Plus 2
I2C:   ready
DRAM:  2 GiB
MMC:   SUNXI SD/MMC: 0, SUNXI SD/MMC: 1
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   phy interface7
eth0: ethernet@1c30000
Hit any key to stop autoboot:  0 
Failed to mount ext2 filesystem...
** Unrecognized filesystem type **
reading boot.bmp
6944 bytes read in 28 ms (242.2 KiB/s)
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
reading /boot.scr
2092 bytes read in 27 ms (75.2 KiB/s)
## Executing script at 43100000
Booting from SD
reading /boot/armbianEnv.txt
** Unable to read file /boot/armbianEnv.txt **
reading armbianEnv.txt
123 bytes read in 14 ms (7.8 KiB/s)
reading /boot/uInitrd
** Unable to read file /boot/uInitrd **
reading uInitrd
2448392 bytes read in 148 ms (15.8 MiB/s)
reading /boot/zImage
** Unable to read file /boot/zImage **
reading zImage
4754648 bytes read in 266 ms (17 MiB/s)
reading /boot/.next
** Unable to read file /boot/.next **
reading .next
** Unable to read file .next **
reading /boot/script.bin
** Unable to read file /boot/script.bin **
reading script.bin
37344 bytes read in 30 ms (1.2 MiB/s)
## Loading init Ramdisk from Legacy Image at 43300000 ...
   Image Name:   uInitrd
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    2448328 Bytes = 2.3 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
Using machid 0x1029 from environment

Starting kernel ...

[sun8i_fixup]: From boot, get meminfo:
        Start:  0x40000000
        Size:   2048MB
[... 368 lines cut ...]
[  204.800037] INFO: Stall ended before state dump start
[  204.780005] Backtrace for cpu 0 (current):
[  204.780005] 
[  204.780005] sending IPI to all other CPUs:
[  204.863325] IPI backtrace for cpu 1
[  204.870007] IPI backtrace for cpu 3
[  204.870005] IPI backtrace for cpu 2
[  240.380052] INFO: task kworker/0:1:24 blocked for more than 120 seconds.
[  240.391166] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  240.403637] kworker/0:1     D c061e084     0    24      2 0x00000000
[  260.410007] INFO: rcu_preempt self-detected stall on CPU
[  260.410007]  0: (1 GPs behind) idle=6dd/140000000000002/0 
[  260.410007] 
[  260.430009] INFO: rcu_preempt detected stalls on CPUs/tasks:
[  260.430023]  0: (1 GPs behind) idle=6dd/140000000000002/0 
[  260.430030]  (detected by 1, t=2002 jiffies)
[  260.430039] INFO: Stall ended before state dump start
[  260.410007] Backtrace for cpu 0 (current):
[  260.410007] 
[  260.410007] sending IPI to all other CPUs:
[  260.497773] IPI backtrace for cpu 1
[  260.500007] IPI backtrace for cpu 3
[  260.500005] IPI backtrace for cpu 2
[  264.030018] BUG: soft lockup - CPU#0 stuck for 22s! [swapper/0:0]
```

Fourdee, 2017-05-25:
@it-praktyk 

Thanks for the report 👍 

Can you please try the following script.bin:
- Write a fresh DietPi image
- Download http://dietpi.com/downloads/binaries/fex_bin/orangepiplus2e.bin
- Save to ```/boot/script.bin```
- Power on system

it-praktyk, 2017-05-25:
I did but without positive effect.

```
U-Boot SPL 2017.01-armbian (Feb 02 2017 - 03:04:04)
DRAM: 2048 MiB
Trying to boot from MMC1

U-Boot 2017.01-armbian (Feb 02 2017 - 03:04:04 +0100) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Xunlong Orange Pi Plus / Plus 2
I2C:   ready
DRAM:  2 GiB
MMC:   SUNXI SD/MMC: 0, SUNXI SD/MMC: 1
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   phy interface7
eth0: ethernet@1c30000
Hit any key to stop autoboot:  0 
Failed to mount ext2 filesystem...
** Unrecognized filesystem type **
reading boot.bmp
6944 bytes read in 29 ms (233.4 KiB/s)
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
reading /boot.scr
2092 bytes read in 28 ms (72.3 KiB/s)
## Executing script at 43100000
Booting from SD
reading /boot/armbianEnv.txt
** Unable to read file /boot/armbianEnv.txt **
reading armbianEnv.txt
123 bytes read in 15 ms (7.8 KiB/s)
reading /boot/uInitrd
** Unable to read file /boot/uInitrd **
reading uInitrd
2448392 bytes read in 149 ms (15.7 MiB/s)
reading /boot/zImage
** Unable to read file /boot/zImage **
reading zImage
4754648 bytes read in 266 ms (17 MiB/s)
reading /boot/.next
** Unable to read file /boot/.next **
reading .next
** Unable to read file .next **
reading /boot/script.bin
** Unable to read file /boot/script.bin **
reading script.bin
38012 bytes read in 30 ms (1.2 MiB/s)
## Loading init Ramdisk from Legacy Image at 43300000 ...
   Image Name:   uInitrd
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    2448328 Bytes = 2.3 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
Using machid 0x1029 from environment

Starting kernel ...

[sun8i_fixup]: From boot, get meminfo:
        Start:  0x40000000
        Size:   2048MB
ion_carveout reserve: 160m@0 256m@0 130m@1 200m@1
ion_reserve_select: ion chipid  [0x2c00081!
[... 341 lines cut ...]
[   88.160007] INFO: rcu_preempt detected stalls on CPUs/tasks:
[   88.160021]  0: (1 GPs behind) idle=35f/140000000000002/0 
[   88.160028]  (detected by 1, t=8007 jiffies)
[   88.160038] INFO: Stall ended before state dump start
[   88.140004]   (t=8011 jiffies)
[   88.140004] Backtrace for cpu 0 (current):
[   88.140004] 
[   88.140004] sending IPI to all other CPUs:
[   88.239453] IPI backtrace for cpu 1
[   88.240005] IPI backtrace for cpu 2
[   88.240006] IPI backtrace for cpu 3
[  148.200004] INFO: rcu_preempt self-detected stall on CPU
[  148.210008] INFO: rcu_preempt detected stalls on CPUs/tasks:
[  148.210022]  0: (1 GPs behind) idle=35f/140000000000002/0 
[  148.210029]  (detected by 2, t=14012 jiffies)
[  148.210038] INFO: Stall ended before state dump start
[  148.200004]  0: (1 GPs behind) idle=35f/140000000000002/0 Έ���6��G���V���z��
[  148.200004]   (t=14017 jiffies)
[  148.200004] Backtrace for cpu 0 (current):
[  148.200004] 
[  148.200004] sending IPI to all other CPUs:
[  148.298660] IPI backtrace for cpu 3
[  148.300005] IPI backtrace for cpu 1
[  148.300007] IPI backtrace for cpu 2
```

Fourdee, 2017-05-27:
@it-praktyk 

Yep, looks broken. Either a hardware failure (try a different image to verify), or, our **OPi Plus** image is now incompatible with **Plus 2**. 

When we created the image for **OPi Plus**, users reported it worked with **OPi Plus 2** (hence the shared image). Although, we do not own a **OPi Plus 2** at this time, so unable to debug/generate a "true" image.

it-praktyk, 2017-05-27:
The board boot when I use armbian. You can try instruct me how we can test/build - I've some Linux knowledge too.

@cbbc can you help?

Fourdee, 2017-05-27:
@it-praktyk 
Yep, you'll need to build a ARMbian Debian Jessie Server image for the OPi Plus 2 (https://github.com/armbian/build). Once done, write to SD card then run through:
https://github.com/Fourdee/DietPi/blob/testing/PREP_SYSTEM_FOR_DIETPI.sh

Once completed, read to image file.

it-praktyk, 2017-05-31:
> Once done, write to SD card then run through (using the testing branch for scripts):

OK, I prepared the Armbian image. Now should I boot Orange Pi from it and the next use the script to "switch" to DietPI?

Fourdee, 2017-06-01:
@it-praktyk 

> OK, I prepared the Armbian image. Now should I boot Orange Pi from it and the next use the script to "switch" to DietPI?

Yep. Don't forget to copy the required DietPi scripts to ```/boot```:
https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L8-L11

Then run through the commands in script above.

it-praktyk, 2017-06-03:
I did it. 

Transcript you can find here: https://gist.github.com/it-praktyk/692d607ab45d26cb4af0321f3e8f6380

Final command can't be found :-/

```
root@orangepiplus2e:~# /DietPi/dietpi/finalise
bash: /DietPi/dietpi/finalise: No such file or directory
```
Yes, I didn't debug it yet - just complain ;-)

Fourdee, 2017-06-03:
@it-praktyk 

> https://gist.github.com/it-praktyk/692d607ab45d26cb4af0321f3e8f6380#file-orangepiplus2e-armbian-to-dietpi-L5827

Looks like DietPi-RAMdisk failed to start. Everything past this point would fail with filepath of ```/DietPi```. We need to see why this failed:
```
systemctl status dietpi-ramdisk.service -l
```

And might be worth checking mount for ```/DietPi``` is up at this point:
```
df -h
```

Fourdee, 2017-06-08:
I've updated all references to remove OPi Plus 2 from the OPi Plus image.

I'll mark this as closed.

@it-praktyk 
If you want to try this again, please continue using this ticket:
https://github.com/Fourdee/DietPi/issues/995

closed 2017-06-08 by Fourdee