MichaIng/DietPi #981: Read-only file system on RPi 3 after finishing the first-run configuration

enolp, 2017-05-27:
I submitted a bug report using dietpi-bugreport, here is the reference code: 4159ed3d-b43c-4d0b-8ef7-6d98f9e533c4-0

I think is not necessary to explain the problem here if I did that bug report, but i forgot to mention in the "Additional notes" section that probably an updated DietPi ISO will solve my issue.

Fourdee, 2017-05-27:
@enolp 

Thanks 👍 

I'll take a look at your bug report and see whats up.

Fourdee, 2017-05-27:
@enolp 

> it suddenly shows read-only file system error breaking Dietpi software

It sounds like a kernel remount to RO, to prevent further damage to file-system. 

You said you tried 2 new SD cards. Can you:
- List the brand of SD card
- List the Power Supply used

enolp, 2017-05-27:
My MicroSD cards brands are:

SAMSUNG 16GB EVO MicroSDHC I Class 10
Lexar 16GB 300x MicroSDHC I Class 10

Finally, the power supply I use is a TONTEC T520LE  input:100-240V-50/60Hz 0.5A output: 5V - 2A

That's strange because when Dietpi starts for the first time, it works perfectly fine an all partitions are resized successful.

Fourdee, 2017-05-27:
@enolp 
I believe this is hardware related (as for what that is, most likely PSU). Unstable system, causing filesystem errors and hence the read only mode to recover.

To exclude SD card's. Try full formatting the SD card, prior to image write (some users report success with this):
https://github.com/Fourdee/DietPi/issues/769#issuecomment-283030267

Failing that, check voltage of PSU output to ensure 4.9v> during load, or try another PSU.

k-plan, 2017-05-27:
> SAMSUNG 16GB EVO MicroSDHC I Class 10

In my experience, this cards, the old black one and the new white/orange are very picky.
I don't like to use this any more.

> Failing that, check voltage of PSU output to ensure 4.9v> during load, or try another PSU.

[The Raspberry Pi Foundation say](https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md), PSU for RPi 3 have to powered by a +5.1V that purchasing a 2.5A 

And voltage drop  on simple micro USB plugs and cable can be a huge problem as well.

enolp, 2017-05-27:
I have a RPi 2 and the RPi 3, so I compared the power supply which they was using and I noted that my RPi 2 was using the RPi 3 power supply, so I reinstalled again Dietpi but same result when updating. I will test Dietpi with Debian Stretch.

@Fourdee I use Linux so I cannot use Windows program unless I create a virtual machine. Any alternatives?

Fourdee, 2017-05-28:
@enolp 

> I use Linux so I cannot use Windows program unless I create a virtual machine. Any alternatives?

You can DD to SD card, same effect. This assumes ```/dev/sda``` is your SD card, change as needed:
```
dd if=/dev/zero of=/dev/sda
```

Fourdee, 2017-05-28:
@k-plan 

> In my experience, this cards, the old black one and the new white/orange are very picky.
I don't like to use this any more.

I had similar issues with the EVO (orange) card on RPi. Filesystem corruptions once or twice. Never used it since, back to Sandisk.

enolp, 2017-05-28:
Well, I ordered an official Raspberry Pi power supply so I hope my problem gets solved. Anyway I'll fill with zeros my Lexar MicroSD because the EVO is in use on the other RPi 2, and update this issue when I have new results.

enolp, 2017-06-02:
UPDATE:

It seems you were right, I received my official Raspberry Pi power supply and after doing some tests, i can conclude all is working fine.

You can close this issue.

Fourdee, 2017-06-03:
@enolp 

Great to hear 👍

closed 2017-06-03 by Fourdee