MichaIng/DietPi #983: DietPi-Config | Enable boot Raspberry Pi 3 from a USB mass storage device

k-plan, 2017-05-28:
Would be nice to bring this feature into `dietpi-config` for easy users experience.

Possible read out `vcgencmd otp_dump | grep 17:` and if it is not set (disabled USB MSD), user can easy change it to enable , so `dietpi-config` it will put in `/DietPi/config.txt` only once and force a reboot like you do, when changing the hostname.

Info: [BOOTMODES > MSD](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md)

Fourdee, 2017-05-29:
Notes:
- @k-plan, confirmed this is a one time setting and non-reversible. ```program_usb_boot_mode=0``` has no effect.

Fourdee, 2017-05-29:
🈯️ Tested on my last RPi 3 that had no bit set.

Completed.

closed 2017-05-29 by Fourdee