MichaIng/DietPi #984: Dietpi pihole orange pi one not fully installing

newdietpi, 2017-05-28:
Install pihole via dietpi on the orange pi one doesn't fully install. Two steps I noticed missing was the setup web interface and either the screen before/after it. Installing via curl works.

Fourdee, 2017-05-29:
@newdietpi 

Thanks for the report 👍 

> Installing via curl works.

Our installation uses the same installer:  http://install.pi-hole.net. However, we use wget:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L4172-L4195

---

> Install pihole via dietpi on the orange pi one doesn't fully install. Two steps I noticed missing was the setup web interface and either the screen before/after it. 

Unable to replicate this in our tests.

- Did you see the following prompt during install?
[image]

If not, and no prompt was displayed regarding "ERROR: URL offline", it could be a 
- freak loss of connection either at your end or http://install.pi-hole.net during the ```wget``` process.
-  http://install.pi-hole.net script was temporarily "broken", and is now fixed.

newdietpi, 2017-05-29:
I replicated the error again last night with a fresh install of dietpi.

I noticed if you select to install pi-hole during the initial dietpi setup vs installing it after it is hit or miss.

I believe it's a combination of the following.

You have to choose not to install pi-hole during the initial dietpi setup then install pi-hole from dietpi afterwards which may work. I was able to get this working one time since I could login via /pihole instead of /admin.
Second time didn't work. I installed pi-hole during the initial dietpi setup, which didn't work. Uninstalled then installed which did via dietpi. Was able login using /pihole
Third time I did the same as the second but used curl which worked.

I think something in the initial setup doesn't fully install pihole?

newdietpi, 2017-05-29:
It's not as smooth installing it on the opi one vs my rpi3 for some reason, I can install it during the initial dietpi everytime on my rpi3 whereas I have to do the above with my opi one

newdietpi, 2017-05-29:
Can you activate newpi on dietpi forums? I had to go on reddit to get help then they pointed me here to report my issue.

Fourdee, 2017-05-29:
@newdietpi 

> You have to choose not to install pi-hole during the initial dietpi setup then install pi-hole from dietpi afterwards which may work. I was able to get this working one time since I could login via /pihole instead of /admin.
Second time didn't work. I installed pi-hole during the initial dietpi setup, which didn't work. Uninstalled then installed which did via dietpi. Was able login using /pihole
Third time I did the same as the second but used curl which worked.

Thanks for the info 👍 
I'll try to duplicate your tests and see if we can find the cause.

closed 2017-05-31 by newdietpi