MichaIng/DietPi #987: Kodi is not listed in software

kGitsov, 2017-05-31:
Hi there,

Installed DietPi recently but I noticed that there is no Kodi included in software, so I run into searching how to include. In DietPi software selection secton is not included, and when I try to install it from terminal it isn't successfully installed and can't run.

What do I have to do?
I'm using Orange Pi PC
DietPi version: 150

Thank you in advance!

Fourdee, 2017-05-31:
@kGitsov 

GPU features are not currently supported on this image. Ideal for server usage.
https://github.com/Fourdee/DietPi/issues/891

Fourdee, 2017-06-01:
@kGitsov 

> Is it possible to upgrade manually with a certain risk? I'm still testing the enviroment :)

By all means, but you'll need to compile Mali driver and Kodi. See links in this post for various methods:
https://github.com/Fourdee/DietPi/issues/891#issue-223369049

Invictaz, 2017-06-07:
@kGitsov Very interested if you can pull this off.

Fourdee, 2018-02-22:
We only support/offer Kodi for RPi/Odroid devices.

closed 2018-02-22 by Fourdee