MichaIng/DietPi #988: DietPi-Software | jRiver

Fourdee, 2017-06-01:
> https://yabb.jriver.com/interact/index.php?topic=95578.0

Fourdee, 2017-06-04:
```
mediacenter22
Segmentation fault
```

---

```
#Requires ALSA XSERVER GPU=128
#dietpi-software install 5 6
apt-get install -y libx11-6 libuuid1 libidn11 libxcb1 libxau6 libxdmcp6 zlib1g libxext6 libgtk2.0-0 libp11-kit0 libstdc++6 p11-kit libgl1-mesa-glx libcanberra-gtk-module xdg-utils xfonts-75dpi xfonts-100dpi lame vorbis-tools musepack-tools xfonts-unifont

wget -q -O - http://dist.jriver.com/mediacenter@jriver.com.gpg.key | apt-key add -
wget http://dist.jriver.com/latest/mediacenter/mediacenter22native.list -O /etc/apt/sources.list.d/mediacenter22.list
apt-get update
apt-get install -y mediacenter22

cat << _EOF_ >> /DietPi/config.txt
framebuffer_depth=32
framebuffer_ignore_alpha=1
_EOF_
```

Fourdee, 2017-06-10:
@sfb2 
Which desktop did you have installed? LXDE?

sfb2, 2017-06-11:
No, XFCE...
"pure" Dietpi..
(try to hold it pure as possible->only Dietpi-suggested software..)
(no further xserver and/or alsa installs and no framebuffer-corrections, only "apt-get install media...")
but so many thx again !! ;-)

Fourdee, 2017-06-11:
@sfb2 

> No, XFCE...
"pure" Dietpi..
(try to hold it as pure as possible->only Dietpi-suggested software..)

Same result on RPi 3. I'll try RPi 2.

sfb2, 2017-06-11:
?
absolute fresh install too..v1.51->see attachements..
(only "SMplayer" is also installed)
[image]
[image]

Fourdee, 2017-06-12:
@sfb2 

We'll i'am stumped lol

> SMplayer xfce-goodies

I'll try fresh installs again and check these are installed 👍

Fourdee, 2017-06-14:
Fresh install RPi 3:
- +XFCE ```dietpi-software install 25```
-
```
wget -q -O - http://dist.jriver.com/mediacenter@jriver.com.gpg.key | apt-key add -
wget http://dist.jriver.com/latest/mediacenter/mediacenter22native.list -O /etc/apt/sources.list.d/mediacenter22.list
apt-get update
apt-get install -y mediacenter22

reboot
```
- 🈯️ Works
- 🈯️ Change GPU split from 64MB to 128MB 

Weird.

Fourdee, 2017-06-16:
@sfb2 

Good to go, thanks for your help on this one 👍 

Marking as completed.

closed 2017-06-16 by Fourdee