MichaIng/DietPi #990: ODROID - hangs with strange symptoms

Hibryda, 2017-06-01:
I have Odroid cloudshell with DietPi. Today I noticed strange situation - it hangs during rsync (or any other mass copy operation) via NFS.
in ```/var/log/netdata/error.log``` I found this:
```
2017-06-02 01:34:10: netdata: ERROR: File /var/cache/netdata/netdata.compression_ratio/savings.db is too old. Clearing it.
2017-06-02 01:34:11: netdata: ERROR: heartbeat missed between 32000126 usec and 32000126 usec
2017-06-02 01:34:11: netdata: ERROR: Number 3503301604838921.0000000 is too big.
2017-06-02 01:34:11: netdata: ERROR: Number -22703359903443004.0000000 is too big.
....
FATAL: All notification methods are disabled. Not sending notification for host 'xxxxx', chart 'net_packets.eth1' to 'sysadmin' for '10s_received_packets_storm' = '2580' for status 'CRITICAL'.
```
What is wrong (yes, packet storm, but why?)

Fourdee, 2017-06-02:
> it hangs during rsync (or any other mass copy operation) via NFS.

I've experienced this when a connection to NFS server is unavailable, it will hang file IO commands. Is the hang longer than 1minute? (thinking HDD spun down on server).

Either way
- Ping the NFS server to ensure its online
- check ```dmesg``` for any clues relating to IO/NFS/NET.
- If CIFS compatible, try this protocol, its more stable during temporary disconnects.

As for the NetData errors, not sure. We do see NetData errors due to unexpected values returned from kernel (bugs), especially on the H3 boards.

Hibryda, 2017-06-02:
@Fourdee The point is that Odroid acts as NFS server in my setup. And it hangs without any clue given except this log (even heartbeat LED stops blinking) while being fed with data. Cannot investigate further as no other logs show anything useful. My board is XU4.

Fourdee, 2017-06-03:
@Hibryda 

> even heartbeat LED stops blinking

Severe hardware/kernel failure if heartbeat is stopping and no info output on HDMI.

Are you running the 6A PSU with cloudshell + XU4? 4A doesn't cut it, especially if HDD attached. I had no end of problems myself and had to upgrade to a 5v/8A PSU for stability.

Failing that:
- does the system hang only when NFS server running?
- might be worth giving the 4.9 kernel image a test run: https://github.com/Fourdee/DietPi/issues/926#issue-226591577

Hibryda, 2017-06-04:
@Fourdee I have 6A PSU as well as even bigger one (22A) for several devices. It's not a matter of power supply - there is no sudden power surge, already checked that.
Yes, the system hangs only on NFS.
As to 4.9 - I use this kernel already due to main fan being unresponsive in lower kernels.

Fourdee, 2017-06-04:
> Yes, the system hangs only on NFS.
As to 4.9 - I use this kernel already 

@Hibryda 

It could be a bug with the 4.9 kernel. Have you tried NFS with 3.x kernel?

Hibryda, 2017-06-08:
Sorry @Fourdee for late response. You are probably right as to kernel error. It started occurring since I upgraded it. But sadly, I need the main fan working - 3.x cannot provide me this comfort. Will try to downgrade when I get a spare.

Fourdee, 2018-04-05:
Marking as closed due to:
- inactivity
- This appears to be a kernel issue with 4.9
- 4.9 kernel is no longer supported/developed by Odroid. 4.14 will be provided with Stretch image

closed 2018-04-05 by Fourdee