MichaIng/DietPi #991: koel installation failed

tatoosh, 2017-06-02:
DietPi v150 to 151 updated and reboot on raspberry pi 2
installed koel via dietpi-software -> got Exception, Screenshot1
http://i.imgur.com/Xmryoij.jpg
Error from Website:
http://i.imgur.com/v31OPX8.jpg

Fourdee, 2017-06-03:
@tatoosh 

Thanks for the report 👍 

Seems SQL is installed fine and DB created. Most likely something went wrong during Koel's installation with ```npm``` and/or ```composer```. 

I'll run some tests here and try to replicate.

In the mean time, can you try uninstalling Koel and installing again?
```
dietpi-software uninstall 143
dietpi-software install 143
```

Fourdee, 2017-06-03:
Fresh install tests:
- 🈯️ Odroid C2:
[image]
- 🈯️ RPi 3 

@tatoosh 
Unable to replicate our end on fresh install. Maybe there is a conflict with other software installed on system. Can you list installed software on your device?
```
dietpi-software list | grep ' =2 '
```

elRadix, 2017-06-03:
can you avoid adjusting  "[Ok] gpumemsplit 64 | Completed" as it is conflicting with kodi where as kodi requires a higher gpu memsplit

sfb2, 2017-06-04:
(sorry, i´m not the master here, as said)
i also guess its a problem with yarn/npm/composer..had in the past almost the same problems,updated to the latest yarn/reinitialized/rebuilded all packages/dependencies->works..

Fourdee, 2017-06-04:
@elRadix 

> can you avoid adjusting "[Ok] gpumemsplit 64 | Completed" as it is conflicting with kodi where as kodi requires a higher gpu memsplit

If Kodi was installed through DietPi-Software, 128MB GPU will be applied, during current software installs and all future ones:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L10477-L10484

tatoosh, 2017-06-07:
@Fourdee:
 dietpi-software list | grep ' =2 '
id 0 | =2 | openssh client:  |
id 1 | =2 | samba client: access network shares | | http://dietpi.com/phpbb/viewtopic.php? dietpi-config > network options: nas/misc
id 5 | =2 | alsa: linux sound system |
id 7 | =2 | ffmpeg: audio & visual libary |
id 9 | =2 | node.js: javascript runtime |
id 16 | =2 | build-essentials: common packages for compile |
id 17 | =2 | git client: git clone etc |
id 32 | =2 | ympd: lightweight web interface music player for mpd | +alsa | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p50
id 37 | =2 | shairport sync: airplay audio player with multiroom sync | +alsa | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1221#p1221
id 69 | =2 | rpi.gpio: gpio interface library for rpi (python) | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=40#p1065
id 80 | =2 | llmp: lighttpd | mysql   | php | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1335#p1335
id 81 | =2 | llsp: lighttpd | sqlite  | php | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1335#p1335
id 84 | =2 | lighttpd: webserver | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1335#p1335
id 86 | =2 | mysql: database | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1335#p1335
id 87 | =2 | sqlite: database | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1335#p1335
id 89 | =2 | php: webserver | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1335#p1335
id 96 | =2 | samba: feature-rich file server | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=10#p56
id 103 | =2 | dietpi-ramlog: minimal, optimized logging | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p68
id 105 | =2 | openssh server:  | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=10#p63
id 118 | =2 | mopidy: web interface music & radio player | +alsa | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=80#p3611
id 128 | =2 | mpd: music player daemon | +alsa | http://dietpi.com/phpbb/viewtopic.php?xxxxxxxxx
id 130 | =2 | python pip: python pip package installer |
id 132 | =2 | aria2: download manager with web interface | +webserver +sqlite | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=6177#p6177
id 138 | =2 | virtualhere: server: share usb devices over the network | | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=6709#p6709
id 143 | =2 | koel: web interface audio streamer | +ffmpeg +nodejs +buildessential +webserver +mysql | http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=7305#p7305

Fourdee, 2017-06-08:
@tatoosh 
Can you try installing the php-sqlite module:
```
apt-get install php5-sqlite
```

If that was missing and installed successfully, please retry the installation:
```
dietpi-software uninstall 143
dietpi-software install 143
```

tatoosh, 2017-06-08:
@Fourdee 
php5-sqlite was already installed.

> Paketlisten werden gelesen... Fertig
> Abhängigkeitsbaum wird aufgebaut.
> Statusinformationen werden eingelesen.... Fertig
> php5-sqlite ist schon die neueste Version.
> 0 aktualisiert, 0 neu installiert, 0 zu entfernen und 4 nicht aktualisiert.
>

Fourdee, 2017-06-16:
@tatoosh 
Apologies for the delay, we are still unable to replicate this. Have you tried installing Koel on a fresh installation of DietPi?

tatoosh, 2017-07-02:
@Fourdee no did not tried it with a fresh reinstall. So you can't reproduce it, it may be a problem at my system. Thanks for helping, i will close the bug.

closed 2017-07-02 by tatoosh