MichaIng/DietPi #992: Wi-Fi devices cannot get a IP address using a RPi 3 as a hotspot and Pi-hole acting as DHCP server

enolp, 2017-06-02:
I'm unable to get IP addresses on my Wi-Fi devices setting a Raspberry Pi 3 as a Wi-Fi hotspot and also, Pi-hole acting as a DHCP server.

ics-dhcp-service fail to load at startup and i tried to mask it to see if that solves my problem but it doesn't. However, I went into dietpi-config in order to configure my Wi-Fi hotspot but there is a limited number of options, and I couldn't disable de DHCP options for the Wi-Fi hotspot.

My idea is, using my RPi 3 as a Wi-Fi hotspot to replace the crappy one which comes with my router and get ad-blocking at the same time.

Bug report: c9ff04fd-1317-4c51-9890-7204c43139f0-0

enolp, 2017-06-03:
Doing a search on the internet to see how can I solve my issue, I found this article: https://frillip.com/using-your-raspberry-pi-3-as-a-wifi-access-point-with-hostapd/ which uses dnsmasq (instead of ics-dhcp-server) like pihole.

I'll do some tests because I think it is not complicated to do and that setup is easy to do.

Fourdee, 2017-06-04:
@enolp 

> Doing a search on the internet to see how can I solve my issue, I found this article: https://frillip.com/using-your-raspberry-pi-3-as-a-wifi-access-point-with-hostapd/ which uses dnsmasq (instead of ics-dhcp-server) like pihole.

This is near exact same as the DietPi-Software: WiFi Hotspot installer.

> Do you think I should to ask this in the Pihole github/reddit?

https://www.reddit.com/r/pihole/comments/4fdcb2/turn_raspberry_pi_into_a_adblocking_wifi_router/

k-plan, 2017-06-05:
@enolp

> I don't know what I did before but now nothing works......

Better to start now from the scratch.

- write a new DiePi image on a sd card

- go through first installation process

- set up via `dietpi-config` eht0 interface with a static ip address, e.g 192.168.0.100, with correct default gateway and a valid dns server 

- enable Onboard WiFi in  `dietpi-config`

- `reboot` your RPi3

- install via ` dietpi-software` WiFi Hotspot

- edit `nano /etc/dhcp/dhcpd.conf` and change `option domain-name-servers`
```
ddns-update-style none;
default-lease-time 600;
max-lease-time 7200;
authoritative;
log-facility local7;

subnet 192.168.42.0 netmask 255.255.255.0 {
        range 192.168.42.10 192.168.42.50;
        option broadcast-address 192.168.42.255;
        option routers 192.168.42.1;
        option domain-name "local";
        option domain-name-servers 192.168.42.1, 192.168.0.100;
}

```

- `reboot` your RPi3

- take a look in `dietpi-config` if  WiFi Hotspot is running. You can as well change SSID, channel, Country Code, secret here.
[image]
[image]

- If WiFi Hotspot is up and running, install `Pi-hole` via  `dietpi-software`. Make sure  what `Webserver Preference  ` is set to ` Lighttpd` before your start ` Install`

- In `Pi-hole` install assistant, choose running interface `eth0`  - not `wlan0` !!

- After finish installation and auto reboot your device, type in terminal: `pihole -a -i all`
Or you can do this via pihole web-interface, e.g.: `192.168.0.100/pihole` => `Login` => `Settings` => `Advanced DNS settings` and choose: `Listen on all interfaces`
[image]

- Imported: And don't activated `Pi-hole DHCP Server`!

- Now you can watch in PiHole `Dashboard` web-interface your connected Wifi HotSpot clients  (IP: 192.168.42.10 - 192.168.42.50) filtered DNS Queries.

It's tested and running here.

----------------------

`ics-dhcp-server` will run for Wifi HotSpot on `wlan0`
[image]

`dnsmasq` is running on `eth0` as DNS Server AND DNS Client at the same time, to get DNS resolution and to feed DNS Queries.
So PiHole `dnsmasq` DHCP Server will as well run on eth0 ...

Fourdee, 2017-06-17:
Marking as closed, please reopen if required.

closed 2017-06-17 by Fourdee

enolp, 2017-06-25:
I almost forgot about this issue but finally I could do some tests.

@k-plan 

Doing that now my ethernet devices can't get an local IP address but my wireless devices get internet access just fine.

I wonder if I have to edit the /etc/dnsmasq.conf file