MichaIng/DietPi #993: List of apt dependencies for dietpi modules | apt-mark auto if possible

MichaIng, 2017-06-02:
It would be great the have a list of all (apt/debian) packages, that are necessary to run the dietpi scripts/commands and for which script/command they are needed.

On top of that, if something is not necessary to run a script/command, but enhances it's functionality in whatever way, this would be also helpful to know. So at long last I would be interested in the reasons for why which package is pre-installed.

I always like to remove as much of my system as possible. I know that dietpi is already one of the most skinny debian packages available. But in my case I additionally can remove all wifi and bluetooth related packages e.g. as I don't use both. So it would be helpful to know, which packages are installed to only provide these functionalities.
There are also some filesystem and archivers related packages, which I don't need.

About many other packages I am not sure, if they are needed by some of the dietpi scripts or just seen as nice-to-have/near-to-necessary for everybody, e.g. all the *utils packages, net-tools, bsdmainutils, pciutils, usbutils, diffutils etc. Some of the provided commands are quite useful, but not necessary to run a slim webserver and no dependencies for the related packages (webserver, database, php, in case ssh). But if some of the dietpi modules/scripts depend on them and will loud or silently fail without, this would be good to know.

A separate dietpi repo with every dietpi module as package with it's dependencies, recommendations and suggestions would be of course successor 😄 .

Last but not least I found many basic/dependency packages marked as manual installed. I like to have just those marked, that I really need, also to have a nice overview and get rid of obsolete packages, as fast as they are not needed anymore due to some upgrade. Especially many libraries are marked as manual installed so far, which can be painlessly marked as auto and will at current stage stay in place after `apt-get autoremove`.

The old and only pull request seemed to go into a similar direction: #632

Fourdee, 2017-06-03:
@MichaIng 

> It would be great the have a list of all (apt/debian) packages, that are necessary to run the dietpi scripts/commands and for which script/command they are needed.

Heres the core list, required to run DietPi (excluding standard system packages in Debian):
https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L126

Bluetooth is installed if the hardware has onboard BT. The packages can be removed if not required:
https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L128-L129

Fourdee, 2017-06-08:
Marking as closed, question has been answered.

closed 2017-06-08 by Fourdee