MichaIng/DietPi #994: [Rpi] raspberrypi specific packages kept back during dist-upgrade

MichaIng, 2017-06-03:
Observed this on Raspberry Pi 2 with dietpi Jessie and Stretch Beta:
```
# apt-get update && apt-get dist-upgrade
...
The following packages have been kept back:
  libraspberrypi-bin libraspberrypi0 raspberrypi-bootloader raspberrypi-kernel
```
Is this intended to avoid firmware related issues and force it's upgrade via dietpi-update/software?

Fourdee, 2017-06-03:
@MichaIng 

Packages are placed on hold, to prevent them overwriting the DietPi RPi kernel (4.9), which is installed by default on RPi's.

Our DietPi kernel is currently the same as per official, however, we enabled 384KHz audio output.

MichaIng, 2017-06-03:
Ah that's true, I also watched repo kernel upgrade 4.4 override kernel 4.9 gained with rpi-update on official Raspbian image 😄.

Fourdee, 2017-06-08:
I'll mark this as closed, question has been answered.

closed 2017-06-08 by Fourdee