MichaIng/DietPi #997: Dietpi Pi3 after update to 151 unable to remote desktop

helio58, 2017-06-06:
trying to "sudo apt-get install -f"
Getting 
"Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 53 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up udev (215-17+deb8u7) ...
Error, MATCH expected but got ' ID_OUI_FROM_DATABASE=Genmark Automation' in '/lib/udev/hwdb.d/20-OUI.hwdb':
Error, key/value pair expected but got ' ID_V' in '/lib/udev/hwdb.d/20-acpi-vendor.hwdb':
Unrecognized character \xAC; marked by <-- HERE after ($ndw_name<-- HERE near column 17 at /usr/sbin/addgroup line 719.
dpkg: error processing package udev (--configure):
 subprocess installed post-installation script returned error exit status 2
Errors were encountered while processing:
 udev
E: Sub-process /usr/bin/dpkg returned an error code (1)"

Any suggestions to solve this? please

Fourdee, 2017-06-08:
@helio58 

> Setting up udev (215-17+deb8u7)
Error, MATCH expected but got ' ID_OUI_FROM_DATABASE=Genmark Automation' in '/lib/udev/hwdb.d/20-OUI.hwdb':
Error, key/value pair expected but got ' ID_V' in '/lib/udev/hwdb.d/20-acpi-vendor.hwdb':
Unrecognized character \xAC; marked by <-- HERE after ($ndw_name<-- HERE near column 17 at /usr/sbin/addgroup line 719.

Looks like a possible dpkg install/script issue with ```udev``` package, or, something wrong with your current ```/lib/udev/hwdb.d/20-acpi-vendor.hwdb```. Not something we've seen previously or currently.

First backup system:
```
dietpi-backup
```

Then try removing the culprit file:
```
rm /lib/udev/hwdb.d/20-acpi-vendor.hwdb
```

Then repair apt (if needed):
```
apt-get install -f
```

And reinstall ```udev```:
```
apt-get install udev --reinstall
```

helio58, 2017-06-08:
Thanks worked :)

closed 2017-06-08 by helio58