MichaIng/DietPi #998: Unable to Install Plex Media Server

MrHaCkEr, 2017-06-07:
At the moment it's not possible to install Plex from the Interface

[image]

Fourdee, 2017-06-07:
@MrHaCkEr 

Thanks for the report 👍 

We'll update the links.

Fourdee, 2017-06-08:
@MrHaCkEr 

Fixed for v152. If you want to install Plex now, please update the following script:
```
wget https://github.com/Fourdee/DietPi/raw/testing/dietpi/dietpi-software -O /DietPi/dietpi/dietpi-software
```

Then install Plex with:
```
dietpi-software install 42
```

closed 2017-06-08 by Fourdee

WolfganP, 2017-06-08:
I assume that the issues were observed in a VM, right? (due the target being amd64)
I have no issues installing plex on RPi2 via the custom dev2day repository.

Fourdee, 2017-06-08:
@WolfganP 

> I assume that the issues were observed in a VM, right? (due the target being amd64)

Yep, the VM installer pulls directly from plex.tv download url. All other devices are apt/repo based.