MichaIng/DietPi #1000: [RPi Stretch] LAAP stack installs "mysql" AND "mariadb" services which disturb each other

MichaIng, 2017-06-08:
I use dietpi stretch to run nextcloud on my RPi2. Chose LAAP webserver stack to assure that mariadb will be used. It is not yet possible to chose preferred database like it is for webserver -> Separate issue already opened about that. Nevertheless depending on that or not, dietpi-services now tries to handle two database services "mysql" AND "mariadb". Actually mysql is not installed, at least not via apt and it does not show up in dietpi-software either. So I can't remove it. dietpi-services starts "mysql" first successfully and tries to start "mariadb" afterwards which fails after long trial time. Nextcloud works fine, using the available service. It IS actually mariadb, which can be seen via modules, version check and nextcloud web ui. I am wondering about this "mariadb" service. DietPi-Process_Tool also shows one entry about that beside : `Webserver MariaDB (1086)` Never saw mariadb running as another service name than "mysql". Did something change there or is it just some typing bug introduced with this dietpi image?

mysql/error.log shows, that indeed the two services seem to disturb each other:
```
2017-06-02 18:22:57 1995976704 [ERROR] mysqld: Can't lock aria control file '/var/lib/mysql/aria_log_control' for exclusive use, error: 11. Will retry for 30 seconds
2017-06-02 18:23:28 1995976704 [ERROR] mysqld: Got error 'Could not get an exclusive lock; file is probably in use by another process' when trying to use aria control file '/var/lib/mysql/aria_log_control'
2017-06-02 18:23:28 1995976704 [ERROR] Plugin 'Aria' init function returned error.
2017-06-02 18:23:28 1995976704 [ERROR] Plugin 'Aria' registration as a STORAGE ENGINE failed.
2017-06-02 18:23:28 1995976704 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2017-06-02 18:23:28 1995976704 [Note] InnoDB: The InnoDB memory heap is disabled
2017-06-02 18:23:28 1995976704 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2017-06-02 18:23:28 1995976704 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2017-06-02 18:23:28 1995976704 [Note] InnoDB: Compressed tables use zlib 1.2.8
2017-06-02 18:23:28 1995976704 [Note] InnoDB: Using Linux native AIO
2017-06-02 18:23:28 1995976704 [Note] InnoDB: Using generic crc32 instructions
2017-06-02 18:23:28 1995976704 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2017-06-02 18:23:28 1995976704 [Note] InnoDB: Completed initialization of buffer pool
2017-06-02 18:23:28 1995976704 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
2017-06-02 18:23:28 1995976704 [Note] InnoDB: Retrying to lock the first data file
2017-06-02 18:23:29 1995976704 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
2017-06-02 18:23:30 1995976704 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
2017-06-02 18:23:31 1995976704 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
2017-06-02 18:23:32 1995976704 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
...
2017-06-02 18:25:07 1995976704 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
2017-06-02 18:25:08 1995976704 [ERROR] InnoDB: Unable to lock ./ibdata1, error: 11
2017-06-02 18:25:08 1995976704 [Note] InnoDB: Unable to open the first data file
2017-06-02 18:25:08 76f83000  InnoDB: Operating system error number 1 in a file operation.
InnoDB: Error number 1 means 'Operation not permitted'.
InnoDB: Some operating system error numbers are described at
InnoDB: http://dev.mysql.com/doc/refman/5.6/en/operating-system-error-codes.html
2017-06-02 18:25:08 1995976704 [ERROR] InnoDB: Can't open './ibdata1'
2017-06-02 18:25:08 1995976704 [ERROR] InnoDB: Could not open or create the system tablespace. If you tried to add new data files to the system tablespace, and it failed here, you should now edit innodb_data_file_path in my.cnf back to $
2017-06-02 18:25:08 1995976704 [ERROR] Plugin 'InnoDB' init function returned error.
2017-06-02 18:25:08 1995976704 [ERROR] Plugin 'InnoDB' registration as a STORAGE ENGINE failed.
2017-06-02 18:25:08 1995976704 [Note] Plugin 'FEEDBACK' is disabled.
2017-06-02 18:25:08 1995976704 [ERROR] Unknown/unsupported storage engine: InnoDB
2017-06-02 18:25:08 1995976704 [ERROR] Aborting

```
dietpi-bugreport Reference CODE: bce03d8e-9e30-406c-95ff-07df36fd1f9e-1

Fourdee, 2017-06-09:
https://github.com/Fourdee/DietPi/issues/894

Fourdee, 2017-06-09:
@MichaIng 

MySQL/MariaDB use the same service name ```mysql```. As DietPi prints the name of the service and process, it will report that the ```mysql``` service/process has been started/applied, even if its actually MariaDB:
```
root@DietPi:~# systemctl status mysql -l
● mysql.service - LSB: Start and stop the mysql database server daemon
   Loaded: loaded (/etc/init.d/mysql)
   Active: active (running) since Fri 2017-06-09 13:32:53 BST; 7s ago
  Process: 2487 ExecStart=/etc/init.d/mysql start (code=exited, status=0/SUCCESS)

root@DietPi:~# mysqld --help
mysqld  Ver 10.0.30-MariaDB-0+deb8u2 for debian-linux-gnu on aarch64 ((Debian))
Copyright (c) 2000, 2016, Oracle, MariaDB Corporation Ab and others.

Starts the MariaDB database server.

```

> mysql/error.log shows, that indeed the two services seem to disturb each other:

Appears bug report did not successfully upload. Can you please try sending it again?
Also, lets check if mysql is installed (```mysql-common``` is part of the mariaDB installation, dependency):
```
dpkg -l | grep mysql
```

---

I was unable to replicate MySQL + MariaDB installs by selecting:
- LLAP
- NextCloud

MariaDB is the only installed SQL DB software and all is functioning as expected.

Fourdee, 2017-06-09:
We do have some code in DietPi software that will prevent MySQL and MariaDB being installed:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L2453-L2476

But this wont for manual installs of mysql outside of DietPi-Software.

MichaIng, 2017-06-09:
> MySQL/MariaDB use the same service name mysql. As DietPi prints the name of the service and process, it will report that the mysql service/process has been started/applied, even if its actually MariaDB:

This is also how I learned it in the past, using the original Raspbian image and this service is up and running well, providing nextcloud database etc.:
```
# systemctl status mysql -l
● mysql.service - LSB: Start and stop the mysql database server daemon
   Loaded: loaded (/etc/init.d/mysql; generated; vendor preset: enabled)
   Active: active (running) since Thu 2017-06-08 17:10:07 CEST; 1 day 3h ago
     Docs: man:systemd-sysv-generator(8)
  Process: 944 ExecStart=/etc/init.d/mysql start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/mysql.service
           ├─ 971 /bin/bash /usr/bin/mysqld_safe
           ├─1124 /usr/sbin/mysqld --basedir=/usr --datadir=/var/lib/mysql --plugin-dir=/usr/lib/arm-linux-gnueabihf/mariadb18/plugin --user=mysql --skip-log-error --pid-file=/var/run/mysqld/mysqld.pid --socket=/var/run/mysqld/mysqld.soc
           └─1125 logger -t mysqld -p daemon error

Jun 08 17:10:05 micha.gnoedi.org mysqld[1125]: 2017-06-08 17:10:05 1995788288 [Note] Plugin 'FEEDBACK' is disabled.
Jun 08 17:10:05 micha.gnoedi.org mysqld[1125]: 2017-06-08 17:10:05 1995788288 [Note] Recovering after a crash using tc.log
Jun 08 17:10:05 micha.gnoedi.org mysqld[1125]: 2017-06-08 17:10:05 1995788288 [Note] Starting crash recovery...
Jun 08 17:10:05 micha.gnoedi.org mysqld[1125]: 2017-06-08 17:10:05 1995788288 [Note] Crash recovery finished.
Jun 08 17:10:05 micha.gnoedi.org mysqld[1125]: 2017-06-08 17:10:05 1995788288 [Note] Server socket created on IP: '127.0.0.1'.
Jun 08 17:10:05 micha.gnoedi.org mysqld[1125]: 2017-06-08 17:10:05 1995788288 [Note] /usr/sbin/mysqld: ready for connections.
Jun 08 17:10:05 micha.gnoedi.org mysqld[1125]: Version: '10.1.22-MariaDB-'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0
Jun 08 17:10:07 micha.gnoedi.org mysql[944]: Starting MariaDB database server: mysqld . ..
Jun 08 17:10:07 micha.gnoedi.org systemd[1]: Started LSB: Start and stop the mysql database server daemon.
Jun 08 17:10:07 micha.gnoedi.org mysql[944]: ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: NO)
```
 But indeed there was also a "mariadb" service established by the webserver stack, which is processed by "dietpi-services" after mysql service and therefore fails, guess because of libraries/databases are already in use (see log at the top):
```
# systemctl status mariadb -l
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; disabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Thu 2017-06-08 17:12:43 CEST; 1 day 3h ago
  Process: 1308 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=1/FAILURE)
  Process: 1219 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
  Process: 1211 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 1200 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 1308 (code=exited, status=1/FAILURE)
   Status: "MariaDB server is down"

Jun 08 17:10:07 micha.gnoedi.org systemd[1]: Starting MariaDB database server...
Jun 08 17:10:08 micha.gnoedi.org mysqld[1308]: 2017-06-08 17:10:08 1995821056 [Note] /usr/sbin/mysqld (mysqld 10.1.22-MariaDB-) starting as process 1308 ...
Jun 08 17:12:01 micha.gnoedi.org systemd[1]: mariadb.service: Start operation timed out. Terminating.
Jun 08 17:12:43 micha.gnoedi.org systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jun 08 17:12:43 micha.gnoedi.org systemd[1]: Failed to start MariaDB database server.
Jun 08 17:12:43 micha.gnoedi.org systemd[1]: mariadb.service: Unit entered failed state.
Jun 08 17:12:43 micha.gnoedi.org systemd[1]: mariadb.service: Failed with result 'timeout'.
```
The "mysqld" related messages seem to mix, as the daemon is used by both processes.

> Appears bug report did not successfully upload. Can you please try sending it again?
> Also, lets check if mysql is installed (mysql-common is part of the mariaDB installation, dependency):

Bug report code: bce03d8e-9e30-406c-95ff-07df36fd1f9e-2
Hope that worked this time. dietpi told me that upload was successful.

```
# dpkg -l | grep mysql
ii  mysql-common                  5.8+1.0.2                   all          MySQL database common files, e.g. /etc/mysql/my.cnf
ii  php7.0-mysql                  7.0.19-1                    armhf        MySQL module for PHP
```

> I was unable to replicate MySQL + MariaDB installs by selecting:
> 
> - LLAP

Just to be precise: I used LAAP and the RPi Stretch image provided: http://dietpi.com/downloads/testing/DietPi_RPi-armv6-(Stretch).7z

It is v145 so the mysql/mariadb patch should just have been included. In between dietpi was upgraded to current v151.

Hmm no idea what went wrong. Do you have an advice how to solve the "problem"? I mean actually everything is running fine, but on every dietpi-services start/restart process it hangs for 1,5 minutes on trying to start/restart this mariadb service. I would just try to disable and remove the related mariadb.service file, but am afraid to break something. According to the apt packages everything is right and no unnecessary mysql package is installed.

Fourdee, 2017-06-10:
@MichaIng 

> mariadb.service
> Just to be precise: I used LAAP and the RPi Stretch image provided: 

Thanks 👍 
Might make all the difference, i'll run some tests on Stretch.

Fourdee, 2017-06-11:
Confirmed:
- Stretch MariaDB install: installs 2 services ```mariadb``` ```mysql```.
- DietPi-Services triggers them both.

---

Stop all services:
- ```systemctl start mariadb``` endless hang. ```"InnoDB: Error: Unable to lock ./ibdata1, error: 11"```

MichaIng, 2017-06-12:
To understand it right:
- The official (Raspbian) Stretch MariaDB package installs both services, maybe as a first independence step for MariaDB or something?
- I see both service names in the `dietpi-services` code: `aSERVICE_NAME` and no excuse, so both will be handled equally.
- Is there some platform where the service `mariadb` is actually needed, as `mysql` in case also triggers MariaDB? Are both just replacements of each other (just for MariaDB of course)? I will have a look at the service files later.
- The other way round: Is there some platform, where the service `mariadb` will not work, if available. In case it is available, one could assume, or could check to be sure, that the package is also installed and used.
- Is there some benefit to use `mariadb` service instead of `mysql`, in case MariaDB is installed? Perhaps it could prevent some problems, if both databases are installed somehow side by side.

Solutions I could think of:
1. Just take out `mariadb` from `aSERVICE_NAME` list (and deactivate it), as/if `mysql` is always available and working together with MariaDB.
2. Check which database is installed (or both; Is this even possible without errors?) and just handle the related service. This only makes sense, if `mariadb` really has some benefit to run MariaDB, compared to `mysql`, otherwise just makes things complicated.

I will get into it later. My system still runs with both services active. So I am available for some tests/checks, if wanted 😃 .

Fourdee, 2017-06-12:
@MichaIng 

> Just take out mariadb from aSERVICE_NAME list (and deactivate it), as/if mysql is always available and working together with MariaDB.

I think thats the solution. 

> Is there some benefit to use mariadb service instead of mysql, in case MariaDB is installed? 

- Appears MariaDB runs fine from both ```mysql``` and ```mariadb``` services.
```
root@DietPi:~# systemctl status mysql
● mysql.service - LSB: Start and stop the mysql database server daemon
   Loaded: loaded (/etc/init.d/mysql; generated; vendor preset: enabled)
   Active: active (running) since Mon 2017-06-12 11:08:36 UTC; 1min 45s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 960 ExecStart=/etc/init.d/mysql start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/mysql.service
           ├─ 991 /bin/bash /usr/bin/mysqld_safe
           ├─1135 /usr/sbin/mysqld --basedir=/usr --datadir=/var/lib/mysql --plugin-dir=/usr/lib/arm-linux-gnueabihf/mariadb18/plugin --user=mysq
           └─1136 logger -t mysqld -p daemon error

Jun 12 11:08:35 DietPi mysqld[1136]: 2017-06-12 11:08:35 1995804672 [Note] InnoDB: Waiting for purge to start
Jun 12 11:08:35 DietPi mysqld[1136]: 2017-06-12 11:08:35 1995804672 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started;
Jun 12 11:08:35 DietPi mysqld[1136]: 2017-06-12 11:08:35 1421034304 [Note] InnoDB: Dumping buffer pool(s) not yet started
Jun 12 11:08:35 DietPi mysqld[1136]: 2017-06-12 11:08:35 1995804672 [Note] Plugin 'FEEDBACK' is disabled.
Jun 12 11:08:35 DietPi mysqld[1136]: 2017-06-12 11:08:35 1995804672 [Note] Server socket created on IP: '127.0.0.1'.
Jun 12 11:08:35 DietPi mysqld[1136]: 2017-06-12 11:08:35 1995804672 [Note] /usr/sbin/mysqld: ready for connections.
Jun 12 11:08:35 DietPi mysqld[1136]: Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0
Jun 12 11:08:36 DietPi mysql[960]: Starting MariaDB database server: mysqld ..
Jun 12 11:08:36 DietPi systemd[1]: Started LSB: Start and stop the mysql database server daemon.
```
- The ```mariadb``` service is systemD: ```/lib/systemd/system/mariadb.service```
- ```mysql``` service is sysinit: ```/etc/init.d/mysql```

Fourdee, 2017-06-12:
@MichaIng 

Just noticed, Issue 1000 👍 :clinking_glasses:

MichaIng, 2017-06-12:
I don't know too much about recognizable differences between systemd and init.d (SysV-Init). Systemd seems to be the newer process manager but also provides managing "old" init.d scripts or something? So `mariadb` seems to be implemented the modern way, enabling systemctl commands etc., init.d scripts on the other hand will run with SysV-Init on older systems? The "service ... start/stop/restart/status" work for both anyway.

Ya I guess in the end it will not make a noticeable difference and just using `mysql` ist the simplest solution without any risk of new bugs through package identification etc. 👍 . Thanks for taking care about that issue 😄 !!

The simple fast fix for me now, before v152 release, would be `service mysql disable`?

Fourdee, 2017-06-12:
@MichaIng 
Tested on both Jessie and Stretch. Fixed for v152. I'll mark this as closed.

> Systemd seems to be the newer process manager but also provides managing "old" init.d scripts or something?

SystemD simplifies the creation of services vastly. And its also capable of threaded service starts (sysinit doesnt support this as i'am aware).

>  So mariadb seems to be implemented the modern way, enabling systemctl commands etc., init.d scripts on the other hand will run with SysV-Init on older systems?

Yep, seems mariaDB now provide a custom service for their package. However, as the mysql service is compatible with both currently, we'll stick with that. Although, as this is stretch, it may change at any time.

> The simple fast fix for me now, before v152 release, would be service mysql disable?

DietPi-Services starts disabled services that match. You'll need to mask the service and prevent it from running:
```
systemctl mask mariadb
```

>  Thanks for taking care about that issue 

No worries, thanks for reporting it 👍

closed 2017-06-12 by Fourdee