MichaIng/DietPi #1002: Dietpi Image for x86_64 computers (not vmware or virtualbox)

lucario1001, 2017-06-08:
now i dont know if there is 

i know there is one for virtualbox and vmware but i have a computer that is just collecting dust is there a iso image for it?

Fourdee, 2017-06-09:
@lucario1001 

> i know there is one for virtualbox and vmware but i have a computer that is just collecting dust is there a iso image for it?

Not possible, due to the vast number of hardware configurations we'd need to support.

DietPi works because we target specific devices and their hardware (eg: a RPi). We can predict and control which hardware will be used on the system. With a PC, that goes out the window.

We do offer a Virtual machine image, which you can run under ESXi. This should give near same performance as native install:
http://www.vmware.com/uk/products/vsphere-hypervisor.html

lucario1001, 2017-06-09:
but is there scripts you can download and use with a debian/ubuntu distro

Fourdee, 2017-06-12:
@lucario1001 

We'll investigate the possibility of providing and supporting a native PC installer:
https://github.com/Fourdee/DietPi/issues/1007

I'll mark this ticket as closed [dupe]. Please use the ticket above for further discussion.

closed 2017-06-12 by Fourdee