MichaIng/DietPi #1007: DietPi on Intel NUC or Gigabyte Brix?

xanview, 2017-06-11:
First of all thank you for making DietPi, it is fantastic and makes life much easier dealing with single board computers. So much so, I'm wondering if I can use it with x86 barebones like https://www.gigabyte.com/Mini-PcBarebone Brix?

I especially like the ability to just flash an image onto a disk and have it auto resize the partition and set everything else up automatically on first login.

Fourdee, 2017-06-12:
@xanview 

Love those mini PC's, had a Acer Revo years back, would love to get my hands on these new beasts :)

We'd had a few requests for a native PC install, i'll take a look and see what we can do. The main issue will be the initial install/image procedure, i'll need to investigate best/simple method to achieve this.

Just bear in mind:
- we won't be able to support GPU features like we do for RPi/Odroid. As the devices will vary. Although, for GPU it will be a simple case of copying a ```xserver.conf``` for the required GPU. Kodi should then run fine.
- Soundcard support will be limited to the default device on system. Should be find on most boards, but unknown results at this time.

Fourdee, 2017-07-01:
Notes:
- /dev/sda required for successful grub boot?

ToDo:
- Lots of testing
- DietPi-Config > Display options: add support for Radeon/Nvidia drivers?

Fourdee, 2017-07-02:
#### My Tests:
- AMD FX 6200
- Nvidia 1050 Ti

DietPi-Software installs:
- 🈯️ LXDE
- 🈯️ Chromium OpenGL

Other:
- 🈯️ DietPi-Config | Nvidia driver install
- 🈯️ DietPi-Drive_Manager: basic mount/unmount tests

Invictaz, 2017-07-04:
@Fourdee Thanks for all the hard work. Will there be a 32 bit distribution so I can run it on an older netbook? And an option to install to harddrive. Otherwise it only runs from USB. I might be able to close the partitions from USB to harddrive but unsure if that will work.

Fourdee, 2017-07-04:
@Invictaz 

>  Will there be a 32 bit distribution so I can run it on an older netbook?

Possibly we might do that in the future. However, with help from our users, test this image first, iron out the bugs then i'll look into it.

Fourdee, 2017-07-04:
@Invictaz 

> And an option to install to harddrive. Otherwise it only runs from USB. I might be able to close the partitions from USB to harddrive but unsure if that will work.

You can write the image directly to a Harddrive, with a USB > Sata converter.

Fourdee, 2017-07-05:
Completed.

Fourdee, 2017-07-05:
Reopened for testing.

Invictaz, 2017-07-05:
@Fourdee but then I have to open a sealed laptop, take out the drive and then write the image to that drive... Bit weird.

Fourdee, 2017-07-05:
@Invictaz 

> but then I have to open a sealed laptop, take out the drive and then write the image to that drive... Bit weird.

Yep, not ideal, but we lack the code currently in DietPi for a USB installer. Depending on how popular the image is, we may look at creating a easy to use USB installer for Native PC image.

Invictaz, 2017-07-05:
I will see if I can test it from a USB stick.

Invictaz, 2017-07-07:
@Fourdee 

I commenced testing on an i7 2670qm laptop.

Has a lot of failed and other errors but seems to work fine. How to send you the bootup log?
Also it's downloading a lot of amd64 stuff when I have an Intel processor... bit weird

The rest is quite fine. How to increase the font size? The resolution back to 720p is a bit weird...

Wifi works. Sound needs to be tested and Bluetooth. Also on the first bootup there were some CPU governor problems but now it's on performance.

ACPI is still a failure and NTPD is slow. Saw also some SSH errors.

Everything's fresh out of the box so only console. 

For now I plan to run without desktop.

Fn + F2 brightness control is working.
Keyboard works
Nvidia drivers are now being installed.

I might turn this thing into a Steam Machine. Will do a feature request later.

For now testing goes on. Will report later if any casualties appear

Cheers

Invictaz, 2017-07-07:
Oh before I forget the images by Splashtop for ASUS are much and much faster. Only 7 seconds to a minimized desktop. Dietpi takes about 35 only to boot to console (on Pi Zero double that time)

Invictaz, 2017-07-07:
Filed the first bug report 

[image]

Also GPU problems in Chrome after installing the Nvidia driver 379.55

[image]

Plus after reboot to Chromium the wifi isn't connecting. I have no serial connection to restart it.

Invictaz, 2017-07-07:
Sound works although not detected in dietpi-config

Fourdee, 2017-07-08:
@Invictaz 

Thanks for report 👍 

Which GPU model are you running?

Invictaz, 2017-07-08:
@Fourdee 

Internal: Intel HD3000
Dedicated: Nvidia GT555M

Fourdee, 2017-07-12:
@Invictaz 

In your BIOS, is there an option to disable the onboard intel HD GPU, and force Nvidia? Might be worth a shot.

Invictaz, 2017-07-13:
@Fourdee No. BIOS is locked on laptops with Optimus dual gpu solution.

Flashing an unlocked bios isn't really an option for me right now. 

From what I know the Intel HD3000 is always used, it offloads tasks to the dedicated Nvidia GPU which renders it and drops the result to the framebuffer of the Intel.

xanview, 2017-07-13:
Probably related to this: https://github.com/Fourdee/DietPi/issues/1066

However with this x86_64 image, DietPi completely kills my openvpn service. Even after reboot, I see:

```
# systemctl status openvpn
● openvpn.service - OpenVPN service
   Loaded: loaded (/lib/systemd/system/openvpn.service; disabled; vendor preset: enabled)
   Active: active (exited) since Thu 2017-07-13 15:46:09 BST; 8min ago
  Process: 3961 ExecStart=/bin/true (code=exited, status=0/SUCCESS)
 Main PID: 3961 (code=exited, status=0/SUCCESS)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/openvpn.service

Jul 13 15:46:09 TestUnit systemd[1]: Starting OpenVPN service...
Jul 13 15:46:09 TextUnit systemd[1]: Started OpenVPN service.
```

Why is it disabled? - I didn't disable it :( - It would be nice to be able to set DietPi up in a where where it doesn't modify services it didn't install/setup.

Fourdee, 2017-07-15:
@Invictaz 

We'll need to do some research: https://wiki.debian.org/NvidiaGraphicsDrivers#Configuration

Mention of BumbleBee project: https://wiki.debian.org/Bumblebee

Invictaz, 2017-07-16:
Yes I saw Bumblebee this week. Might test it later.

Also there is the option (somewhere in Linux) to disable the dedicated graphics and run only on Intel HD graphics. This saves energy which is one of the Key aspects of DietPi. So might need to be included.

Invictaz, 2017-07-16:
@Fourdee Bumblebee is last updated in 2013. We might use the proprietary Nvidia Driver from non-free plus the newest Xorg 1.19. I see in various bug reports of DietPi that 1.18 is used.

Also there is no indication in dietpi-config which Nvidia Driver version is installed. I believe its 379.55 but not really sure. It needs nvidia_drm.modeset to have PRIME synchronization to remove the tearing

http://www.phoronix.com/scan.php?page=news_item&px=Xorg-PRIME-Sync-Double-Buffer
https://devtalk.nvidia.com/default/topic/957814/prime-and-prime-synchronization/?offset=169
https://askubuntu.com/questions/900286/with-nvidia-optimus-xorg-1-19-3-kernel-4-10-0-15-and-still-having-screentear
https://www.gamingonlinux.com/articles/how-to-an-update-on-fixing-screen-tearing-on-linux-with-an-nvidia-gpu.8892/page=3

We might also have the 32bit driver needed if Steam is coming in Dietpi v154 or v155

https://solus-project.com/forums/viewtopic.php?t=2029

Fourdee, 2017-07-17:
Issues with NV audio, always reverts HDMI output to mute on reboot:
[image]

sal666, 2017-07-26:
As for how to write the image to the HD, can I suggest [Clonezilla](http://clonezilla.org)?
You don't need to open your laptop/PC, you just boot out of a USB/CD and flash the image.
[Here](http://clonezilla.org/liveusb.php) you can find instructions to make a bootable USB flash drive.
Basically it all boils down to:
1. Unzip Clonezilla to the USB flash drive.
2. Copy the DietPi image to the USB flash drive.
3. Tweak two text files (one for UEFI boot, one for BIOS boot).
4. Only for BIOS boots, run a script already in the flash drive so syslinux is installed.

I have some experience with that. Drop me a line if I can be of any help.

belabelabela, 2017-08-09:
@sal666 
I have tried to tweak the two text files as you mentioned in step 3. It seems I am not experienced enough, because it did not work for me. :(

Could you describe the necessary steps (3-4) in detail?

Thanks

Fourdee, 2017-08-10:
### Intel GPU driver install (tested with HD520, should work for all):
NB: No Xserver must be running.
- ```apt-get install -y xserver-xorg-video-intel libgl1-mesa-dri
Xorg -configure
mv $HOME/xorg.conf.new /etc/X11/xorg.conf
```

Invictaz, 2017-08-10:
Thanks! It is either NVIDIA or Intel in Optimus situation. Best advice is if you have such a dual GPU setup run everything on the NVIDIA while this copies everything into the Intel framebuffer (which is connected to LVDS display). 

If you have only an Intel card (or don't care about the additional performance of the NVIDIA) do as @Fourdee suggested.

sal666, 2017-09-08:
@belabelabela 

For step 3, refer to GRUB documentation: 
[-GRUB](https://www.gnu.org/software/grub/grub-documentation.html)
[-tips about theming](https://wiki.archlinux.org/index.php/GRUB/Tips_and_tricks#Manually_creating_grub.cfg)
[-more about theming](http://wiki.rosalab.ru/en/index.php/Grub2_theme_tutorial)

Step 4 is explained [here (step #5)](http://clonezilla.org/liveusb.php#linux-method-b)

I've prepared a quick demo. Do the following:
1. Install Clonezilla live to your USB flash drive (full instructions [here](http://clonezilla.org/liveusb.php)):
    - download alternative stable version from [here](http://clonezilla.org/downloads.php).
    - extract it to your pendrive.
2. Make your pendrive bootable [(step #5)](http://clonezilla.org/liveusb.php#linux-method-b).
3. Unzip this file to your pendrive: [DietPi-Clonezilla.zip](https://github.com/Fourdee/DietPi/files/1287962/DietPi-Clonezilla.zip).
4. Edit grub.cfg, isolinux.cfg and syslinux.cfg and replace 'sda' with the name of your target hard disk.
5. Copy Clonezilla's image files to <your_pendrive>/home/partimag/DietPi/
6. Boot from your pendrive.
7. You should see this boot screen:
[image]
8. **WARNING**: Choosing "Install DietPi" will restore DietPi to your first hard disk (sda or whatever your chose at step 4).

Fourdee, 2017-09-09:
@sal666 

This is awesome, great work and thanks for sharing 👍

belabelabela, 2017-09-10:
@sal666

Thanks a lot, I will give it a try!

Fourdee, 2018-01-06:
Marking as closed. Image is now available for MBR and EFI.

closed 2018-01-06 by Fourdee