MichaIng/DietPi #1011: Git | Create Pull request / issue request templates

Fourdee, 2017-06-12:
> https://github.com/blog/2111-issue-and-pull-request-templates
https://github.com/Fourdee/DietPi/pull/1008#issuecomment-307788094

Fourdee, 2017-06-12:
https://github.com/Fourdee/DietPi/blob/testing/PULL_REQUEST_TEMPLATE.md

noplanman, 2017-06-12:
If I may leave a suggestion, it makes sense to make the text in the template an HTML comment, so that the text itself wouldn't get added to the PR comment if somebody leaves it there.

Example here: https://github.com/php-telegram-bot/core/tree/master/.github

(Also, you can move it to a `.github` folder to keep it stored nicely separately)

Fourdee, 2017-06-13:
@noplanman 

> HTML comment, so that the text itself wouldn't get added to the PR comment if somebody leaves it there.

Nice 👍 

I'll be borrowing your ```.md``` templates 😉

noplanman, 2017-06-13:
> I'll be borrowing your .md templates 😉

The beauty of open source 😇

(DietPi FTW by the way! Loving the simplicity and lightweightness 😍)

Fourdee, 2017-06-16:
https://github.com/Fourdee/DietPi/blob/testing/ISSUE_TEMPLATE.md

Fourdee, 2017-06-16:
@noplanman 

Many thanks again for your help on this 👍 

I'll mark this as completed, should start seeing this active once v152 is released (merged to master)?

Fourdee, 2017-06-16:
Lets put them in the .github folder

Fourdee, 2017-06-16:
https://github.com/Fourdee/DietPi/tree/testing/.github

closed 2017-06-16 by Fourdee