MichaIng/DietPi #1017: DietPi-Software | Deluge (update request)

FireHelmet, 2017-06-14:
Hi all,

Is it possible to update Deluge to the latest version (1.3.15) at the installation step ?

Currently, the version is 1.3.10 (october 2014 !!!)

Thanks !

Fourdee, 2017-06-15:
@FireHelmet 
Deluge is currently a apt-get installation. I'll see if we can switch to source build (depending on build time):
http://dev.deluge-torrent.org/wiki/Installing/Source

Fourdee, 2017-12-30:
Marking as closed, we'll continue to use apt installations + Stretch.

closed 2017-12-30 by Fourdee

gotttlieb, 2018-01-17:
Ok, i see that this thread is close, but version 1.3.10 has a lot of nasty bugs with the Proxy settings I can't set my there.
And seems that XU4 is still not Stretch, so, it seems that we should need to manually fix this?

MichaIng, 2018-01-18:
@gotttlieb 
I found a PPA, which Debian users successfully used in the past. Try:
`echo 'deb http://ppa.launchpad.net/deluge-team/ppa/ubuntu xenial main' > /etc/apt/sources.list.d/deluge.list && apt update && apt dist-upgrade`

If this works well, we can consider to use in on DietPi Jessie.

Odroids are still on Jessie yes. We use Meverics image as base: https://forum.odroid.com/viewtopic.php?f=96&t=17542
And there are still issues with Kodi and other software we provide, on using the experimental Stretch image/repo. Once these are solved, we will immediately switch our Odroid images as well.

gotttlieb, 2018-01-18:
@MichaIng I can confirm that now Deluge is working as expected. 
I really envourage this change in the images, at least in the Odroid ones.
Thanks so much for your support