MichaIng/DietPi #1019: DietPi-Software | Standard Notes

joaofl, 2017-06-16:
Nice Evernote-like application. It is open-source, encrypted, multi platform (Linux, windows, mac, iOS, Android), and allows users to run their own server instance.

Requires Ruby, MySQL, and a web server.

Links:
https://standardnotes.org/
https://github.com/standardnotes

I'll give it a try, install it, and could eventually try to add it to DietPi...
@Fourdee could you give some tips on that?

Fourdee, 2017-06-17:
@joaofl 

> I'll give it a try, install it, and could eventually try to add it to DietPi...

Excellent 👍 

Yep, created a guide to adding software in DietPi-Software here: 
https://github.com/Fourdee/DietPi/issues/490#issuecomment-244416570

If you get into any issues, or unsure, let me know and i'll assist.

joaofl, 2017-06-19:
@Fourdee 
I was trying to manually install it before attempting to make any kind of script, and It is a bit tricky actually.

Starting by the fact that it requires Ruby2.2+, while v2.1 is the one available in our repos. But with a few command lines we can actually download and compile Ruby2.4, which solves the issue.
Further on, the installation fails again, which seems to be caused by some MySQL version incompatibility.

I'm still not sure about the best way to solve those issues, but I hope to figure out soon.

joaofl, 2017-06-19:
@Fourdee 
They actually require at least MySQL 5.6, while we have v5.5. I'm afraid this upgrade may cause too many headaches. What do you think should be done? 

from: https://github.com/standardfile/ruby-server/issues/48#issuecomment-309468373

joaofl, 2017-06-21:
After Upgrading MySQL to the latest, and every other requirement, I managed to get the server up, but still, not satisfactorily. It was crashing, or either giving errors on registering new user.

The apps also seem to be a bit premature (both desktop and iphone).

Because of that, I went on a detailed search for apps like these, and found some alternative like:
Turtl, Papertwork, Laverna. I've managed to put Turtl and Laverna servers running, but each one with its own limitations: either too complicated to set up the server, huge dependencies, or very limited functionalities.

For now I'll give up on it, and give another try further on... Should I close the issue and reopen when I get something?

Fourdee, 2017-06-22:
@joaofl 

> The apps also seem to be a bit premature (both desktop and iphone).

Usually a good indicator to avoid this software until it improves over time.

> Should I close the issue and reopen when I get something?

Entirely your call. I tend to leave started/unfinished tickets in the prototyping branch and open. Although, if you believe this software isn't worth the integration into DietPi, due to the current issues, close her up :)

closed 2017-12-30 by Fourdee