MichaIng/DietPi #1023: DietPi-Software | Mono fails install for ARMv6

Fourdee, 2017-06-19:


Fourdee, 2017-06-19:
- Find another source for compatible ARMv6 mono installation packages. Currently ```deb http://download.mono-project.com/repo/debian jessie main```

Fourdee, 2017-06-20:
- 🈺 RPi Zero: Installing all 3 software titles (```dietpi-software install 144 145 147```) at the same time results in ```/tmp``` running out of space:
```
tar: A lone zero block at 42380
awk: write failure (No space left on device)
awk: close failed on file /dev/stdout (No space left on device)

root@DietPi:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           241M  241M     0 100% /tmp

```

Fourdee, 2017-06-22:
I'll mark this as closed, now resolved.

New ticket for outstanding issue with ```/tmp``` 0 space:
https://github.com/Fourdee/DietPi/issues/1027

closed 2017-06-22 by Fourdee

MichaIng, 2018-08-27:
@Fourdee 
This resolves https://github.com/Fourdee/DietPi/issues/534 as well, right?