MichaIng/DietPi #1026: Line Out AlsaMixer defaults to off

kurtjcu, 2017-06-21:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version= 152
- SBC Device Orange Pi zero + Orange Pi expansion board
- Power supply used 5v 3a PS?
- SD card used Sandisk Ultra
- Distro = Linux DietPi 3.4.113-sun8i #18 SMP PREEMPT Thu Jun 15 02:16:06 CEST 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:
The Lineout should be set to "on" after choosing 3.5mm analog audio via the wizard.

#### Actual behaviour:
With a fresh install and enabling alsa (then shairport) and setting to 3.5mm analog audio as the default output. The line out in alsa mixer is set to "off" by default. 

To correct this I typed "amixer cset numid=9 on" to enable the analog output for alsa.

#### Steps to reproduce:
Fresh install of Dietpi, select 3.5mm analog audio in wizard.

type in "alsamixer" in the terminal and note that the 6th control from the left "audio Lineout" is set to "off"

#### Did you submit a ```dietpi-bugreport```?
no, where do i do that?

#### Extra details:

sfb2, 2017-06-22:
nice description !! (can´t help from here, but nice written ;-) )

Fourdee, 2017-06-22:
@kurtjcu 
Many thanks for the report 👍 

We'll run tests here and try to replicate.

Fourdee, 2017-06-24:
@kurtjcu 

Confirmed, thanks for letting us know 👍 

We'll get this fixed for v153 and mute/unmute lineout as needed.

Fourdee, 2017-06-24:
Completed with commit: https://github.com/Fourdee/DietPi/commit/99eb7527cbadd3984f4a84f0b86c8765d8dcbfb2

closed 2017-06-24 by Fourdee

kurtjcu, 2017-06-30:
Thanks for the resolution :) Im really getting into diet pi 👍