MichaIng/DietPi #1027: DietPi-Software | /tmp out of free space

Fourdee, 2017-06-22:
In DietPi ```/tmp``` is mounted to RAM with default size of 50% of RAM.

In some instances (https://github.com/Fourdee/DietPi/issues/1023#issuecomment-309731931), software installations on 512MB < RAM will run into this issue.

We need to either:
- Increase size of ```/tmp``` from default to 80%
- Disable ```/tmp``` RAM mount on 512MB or lower devices.

Fourdee, 2017-06-23:
Completed with solution:
- Disable /tmp RAM mount on 512MB or lower devices.

Applied during patch and during 1st run of DietPi.

Fourdee, 2018-02-28:
Due to: https://github.com/Fourdee/DietPi/issues/1566#issuecomment-369296662

Fourdee, 2018-02-28:
Ref: https://github.com/Fourdee/DietPi/commit/cd093b085fcdb3e84df975e80b75a4633a6539ca

Fourdee, 2018-02-28:
2 options:
- Force a size of ```/tmp``` to ```1.5GB```, which with our standard 2GB-RAM swapdisk size, should be fine and allow for swap
- Disable ```/tmp``` as ```tmpfs``` with devices on 1GB or less memory.

k-plan, 2018-02-28:
@Fourdee 

> - Disable `/tmp` RAM mount on 512MB or lower devices.

Have mentioned personally dislike to do this some times ago.
We give away some sort of protection for sd card life time.

Question (without skill):

We now "build" swapfile and size directly before install software.
Is it not possible to put downloaded mono installation direct to `swap` and then install / extract it?
So we can keep `tmpfs` for all devices?

Fourdee, 2018-02-28:
@k-plan 

> Is it not possible to put downloaded mono installation direct to swap and then install / extract it?
So we can keep tmpfs for all devices?

Cant see a way of changing the APT/DPKG directory used during extraction. But Option 1 would cover all devices and allow ```/tmp``` as ```tmpfs``` https://github.com/Fourdee/DietPi/issues/1027#issuecomment-369346584

Reference, mono install:
```
root@DietPi:~# G_TREESIZE /tmp
501.4 MB /tmp

# many of these
-rw-------  1 root root 123M Feb 28 18:02 mono_aot_EhsfHq
```

k-plan, 2018-02-28:
> - Disable /tmp as tmpfs with devices on 1GB or less memory.

:cry:  :fearful: 

option 3:

- put mono and all depended software into bin ...

k-plan, 2018-02-28:
> - Force a size of /tmp to 1.5GB, which with our standard 2GB-RAM swapdisk size, should be fine and allow for swap

Hmm ... question is, at which point this can be done?
- Default on all images while initial bootup and create a "standard" swap file of 1.5GB and  `/tmp `RAM mount on 512MB?
- Or can it be done, ONLY before `mono` has selected to be installed? (dynamic)

Fourdee, 2018-02-28:
@k-plan 

> question is, at which point this can be done?

During 1st run of boot script. But i'll patch this for v6.3 and existing systems if we go ahead.

So we could:
- If RAM < 2GB, and swapfile size is default (2GB-RAM), we could force a size of 1.0/1.5?GB /tmp.
- If RAM >= 2GB, we could leave the default tmpfs size (which is 50% of total RAM). 2GB RAM = 1GB ```/tmp```

k-plan, 2018-02-28:
Question 2:

> Reference, mono install:
> 
> ```
> root@DietPi:~# G_TREESIZE /tmp
> 501.4 MB /tmp
> 
> # many of these
> -rw-------  1 root root 123M Feb 28 18:02 mono_aot_EhsfHq
> ```
> Reference, mono install:
> 
> ```
> root@DietPi:~# G_TREESIZE /tmp
> 501.4 MB /tmp
> 
> # many of these
> -rw-------  1 root root 123M Feb 28 18:02 mono_aot_EhsfHq
> ```

Will this mono installation :hankey:  be deleted after installation?

Fourdee, 2018-02-28:
@k-plan 

> Will this mono installation 💩 be deleted after installation?

lol yep, After reboot

Fourdee, 2018-02-28:
@k-plan 

> If RAM < 2GB, and swapfile size is default (2GB-RAM), we could force a size of 1.0/1.5?GB

Example, on a 512MB RAM system:
- 1.5~GB swap
- 1GB ```/tmp```
- Lets say all RAM is used by processes (not ```/tmp```), start of swapping
- ```/tmp``` has upto 1GB available still via swap.
- if ```/tmp``` is full, swap should still have 500MB left over.

In other words, we should be fine setting a higher ```/tmp``` size than RAM, as long as the swapfile is large enough to accommodate it, should swapping occur.

MichaIng, 2018-03-01:
@Fourdee 
Jep agree, we could just set /tmp to 50% of RAM+swap instead if just RAM size. It is a shame, that dpkg does not handle this unpacking into /tmp more efficient, e.g. unpack single package, afterwards remove files laid out on/tmp, before unpacking the next. I mean /tmp just fills, because of the huge amount of dependencies of `mono-devel` etc.
I also searched the web quite a while for a possibility to change the unpack directory, but sadly found nothing, not within apt, nor dpkg config files/options, nor setting $TMPDIR (used by dpkg for tmp files/folders, but during my tests not for the unpacked packages...).

As this could happen on every large installation or if users decide to lower/disable their swap file, maybe we could catch this specific error during G_AGI and inform users, to increase their swap size?

Fourdee, 2018-03-01:
notes:
- We must apply this during swapfile apply. As we rely on the swapfile size to be 2GB-RAM = total.

Fourdee, 2018-03-01:
```
root@DietPi:~# /DietPi/dietpi/func/dietpi-set_dphys-swapfile 0
[  OK  ] Root access verified.

 DietPi
─────────────────────────────────────────────────────
 Mode: Disable swapfile

[ INFO ] DietPi-dphys-swapfile | Deleting existing swapfile (/var/swap)

[ INFO ] DietPi-dphys-swapfile | Setting /tmp tmpfs size: 488M
[  OK  ] DietPi-dphys-swapfile | mount -o remount,size=488M tmpfs /tmp
[ INFO ] DietPi-dphys-swapfile | Completed

root@DietPi:~# /DietPi/dietpi/func/dietpi-set_dphys-swapfile 500
[  OK  ] Root access verified.
[ INFO ] DietPi-dphys-swapfile | Size     = 500 MB
[ INFO ] DietPi-dphys-swapfile | Location = /var/swap
[ INFO ] DietPi-dphys-swapfile | Deleting existing swapfile (/var/swap)

want /var/swap=500MByte, generating swapfile ... of 500MBytes
[ INFO ] DietPi-dphys-swapfile | Setting /tmp tmpfs size: 738M
[  OK  ] DietPi-dphys-swapfile | mount -o remount,size=738M tmpfs /tmp
[ INFO ] DietPi-dphys-swapfile | Completed

root@DietPi:~# /DietPi/dietpi/func/dietpi-set_dphys-swapfile 1
[  OK  ] Root access verified.
[ INFO ] DietPi-dphys-swapfile | Size     = 1072 MB
[ INFO ] DietPi-dphys-swapfile | Location = /var/swap
[ INFO ] DietPi-dphys-swapfile | Deleting existing swapfile (/var/swap)

want /var/swap=1072MByte, generating swapfile ... of 1072MBytes
[ INFO ] DietPi-dphys-swapfile | Setting /tmp tmpfs size: 1024M
[  OK  ] DietPi-dphys-swapfile | mount -o remount,size=1024M tmpfs /tmp
[ INFO ] DietPi-dphys-swapfile | Completed
```

k-plan, 2018-03-01:
@Fourdee 

Thanks for declaration. Yes, I have imagine. it will work like this.

provocative question from a user point of view:

- have e.g. RPi Zero (512GB RAM) and want smal (diet) installtion for a single use case , e.g. IOT, Pi-Hole, CAM, ...
- For what I do need a 1.5GB swap file, if I never install `mono`?
- Why do my installation take soo long and my device will not react and green led will light? With former version I never have this problem.
- Why can I not  use my old 2GB sd card, which run without problem for years now with DietPi v.15x? My full installation is only about 600MB?

--------------------

> Fourdee:
> But i'll patch this for v6.3 and existing systems if ...

> MichaIng:
>  .... if users decide to lower/disable their swap file, maybe we ....

Good points and they point into the direction I like go:

Idea / approach : 

- if we can patch every running installation, why we don't do this, only if user need / want e.g.` mono` to 2GB-RAM `swap file` and  `/tmp` 
- after check-in some software, which need such a huge amount of space e.g. `mono` 
- check if enough free space is available on file system 
- we have to notice User via message, what  will happen now, before installation will start or break it, if where is not enough free space with message
- write back a run-once service for  check-in software installation after next boot or login. If possible, as 
run-once Automatic root login.
- reboot the device
- patch the device 
- Automatic login as root as run-once service
- swap file creation to 2GB-RAM
- start selected software installation as run-once service, like `dietpi.txt` will do during initial setup
- after installation is finished, remove all  run-once service before auto reboot
- auto reboot to finish software installation

If we can get this to run:

- we can withdraw https://github.com/Fourdee/DietPi/issues/1027 to "old stile RAM /tmp file system via version patch  now
- we can withdraw https://github.com/Fourdee/DietPi/issues/1130#issuecomment-350351430  via version patch now
- we have improved live span of fault-prone a little
- we keep base installation diet and small, like "DietPi" implicit 
- we avoiding file system fault while initial boot up (huge power consumption with voltage drop)

Fourdee, 2018-03-01:
Completed.

Fourdee, 2018-03-01:
Actually no:
```
/DietPi/dietpi/func/dietpi-set_dphys-swapfile $(grep -m1 '^AUTO_SETUP_SWAPFILE_SIZE=' /DietPi/dietpi.txt | sed 's/.*=//')
```

Will set current swapfile size, instead of auto as the var is updated to actual value once done. We may need to force a reset of auto swapfile size as we have no way of knowing if that was originally applied.

closed 2018-03-01 by Fourdee

reeslo, 2020-03-18:
Guys/Girls how to that after long time install and /tmp full ?

MichaIng, 2020-03-18:
@reeslo 
Please check what fills your /tmp. That should only be used temporarily for relatively small file sizes. Or is there any specific install that fails, which writes lots of files to /tmp?
```
G_TREESIZE /tmp
```

reeslo, 2020-03-18:
I thanks for reply, I found that was the swap file. And I do this https://dietpi.com/phpbb/viewtopic.php?p=18433#p18433 but seem not fix my problem, my pi is not responding.. I have to investigate, its looks like other issue. :/

reeslo, 2020-03-18:
I have send a bug remport `Bug report sent, reference code: c303265e-fe6f-4af8-890c-e21620b589b9`

MichaIng, 2020-03-19:
@reeslo 
I checked your bug report. Indeed every looks pretty fine, swap file is there, nearly unused, even physical RAM mostly free.

I see you have wpa_supplicant.service enabled, without having an WiFi card, and even if, ifupdown invokes wpa_supplicant ondemand per-interface. Hence you should disable it: `systemctl disable wpa_supplicant`
Another minor thing to satisfy Dropbear about a missing log file: `> /var/log/lastlog`

What is the actual issue? What you mean by Pi is not responding, as you are able to login and send a bug report at least?

reeslo, 2020-03-19:
@MichaIng thanks for reply,  my pb was pi not responding, after reboot I see swap full, then I change swap file destination, but without reboot, pi not responding after hours.. I reboot one more time, and at now pi is up  (16 hours). If I have anymore my pi not responding I make investigation, like your suggestions.

Thanks and sorry for my english 😬

reeslo, 2020-03-20:
@MichaIng there is no more freezer (2 days ultime), I think I've solved the issue by changing swap file location to /root/my_swap ! 

[image]