MichaIng/DietPi #1028: Image request for the Up Board

ifvictr, 2017-06-23:
Thanks for spending your time on DietPi, I really love how all my boards are unified using a common OS. Unfortunately, I couldn't seem to find an image for the [Up Board](http://www.up-board.org/up/) anywhere on the DietPi website. Would you be able to create an image for it? Thanks!!!

Fourdee, 2017-06-23:
@ifvictr 

Hi, thanks for the request 👍 

Been keeping an eye on the Up boards, look really impressive. However, still on the wish list at the moment.

As this board will run native x86_64 linux, this should tie in with the following generic x86_64 native image ticket request:
https://github.com/Fourdee/DietPi/issues/1007

I'll bump the above ticket and take a look for v153.

Notes:
- Excellent, appears upboards have a Bios: https://up-community.org/forum/general-discussion-up/1197-cloning-the-image-of-an-up-board#3810

Invictaz, 2017-06-23:
Interesting, an Atom based board. 

If possible, can this also run on Atom based phones and tablets or does that need a bootloader modification?

Fourdee, 2018-04-05:
Marking as closed due to:
- This ties in with our native PC image and should be functional: https://github.com/Fourdee/DietPi/issues/1171#issuecomment-336522021

closed 2018-04-05 by Fourdee