MichaIng/DietPi #1030: Hostname not broadcast

GrahamGIGA, 2017-06-24:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?

152

- SBC Device (EG: RPi 3)?
Rpi 3
- Power supply used (EG: 5v 1A RAVpower)?
USB

- SD card used (EG: Sandisk ultra)?

Sandisk
- Distro (EG: Jessie) | ```uname -a```?
Jessie

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

This is fresh :-)

#### Expected behaviour:

I changed hostname (RoonKitchen and Roon_Kitchen) and it did not display when scanning for IPs

#### Actual behaviour:

Hostname is blank

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2017-06-25:
@GrahamGIGA 

Thanks for the report 👍 

I was unable to replicate this on RPi3, setting ```Roon_Kitchen``` as hostname in ```dietpi-config``` > Security options > Hostname, then rebooting, hostname is applied correctly.

The only thing that may throw this out is a space at the start of the hostname, ensure this doesn't exist.

> Hostname is blank

Is this blank in:
- ```dietpi-config``` > Security options > Hostname ?
- ```cat /etc/hostname``` ?
- ```cat /etc/hosts```?

If its not blank, it may be router is not updating the hostname of device, possibly due to current DHCP lease. Try restarting the router to see if it refreshes the new hostname of device.

GrahamGIGA, 2017-06-30:
Hi Dan, I'll give that a try!

Cheers

GrahamGIGA, 2017-07-03:
Hi Dan, that fixes the publication.  However one odd issue if I name the host Roon_Boom (or anything else with an underscore) the underscore is removed when rpeorting the Hostname.

Cheers

Graham
[image]

k-plan, 2017-07-04:
@GrahamGIGA 

>  However one odd issue if I name the host Roon_Boom (or anything else with an underscore) the underscore is removed when rpeorting the Hostname.

No,not DietPi cause this, it is a restrictions on valid hostnames corresponding to [RFC](https://en.wikipedia.org/wiki/Request_for_Comments):
[Restrictions on valid hostnames](https://en.wikipedia.org/wiki/Hostname#Restrictions_on_valid_hostnames)

@Fourdee 

it will be a nice feature, to put `avahi-daemon` -  mDNS, Zeroconf, Bonjour - how ever you call it, as option into `DietPi-Software | Add additional linux software options`. So we can install it automated.
May be helpful for music-player demons, samba, print server, RPi Zero Ethernet Gadget and so on ...

Fourdee, 2017-07-04:
@k-plan 

Great suggestion 👍 

I'll add to DietPi-Software. We could also include it as a pre-req for all audio software installations.

k-plan, 2017-07-04:
@Fourdee 

> We could also include it as a pre-req for all audio software installations.

But it will help only Mac-user. Windows doesn’t have Zeroconf/Bonjour support out of the box.
Yes, you can install it: https://learn.adafruit.com/bonjour-zeroconf-networking-for-windows-and-linux/#microsoft-windows

Audio software installations, especially with streaming, mostly use services like Universal Plug and Play (UPnP), Airplay or Digital Living Network Alliance (DLNA) to broadcast self announce.
Most of them do this out of the box.

Fourdee, 2017-07-07:
Avahi-daemon is currently prereq for (automatically installed as needed):
- MPD
- Shariport-sync
- Virtual Here

Aside from the above (where its required for the software), I believe its best we leave the installation of Avahi-Daemon optional by end-user.

I'll mark this as completed.

closed 2017-07-07 by Fourdee