MichaIng/DietPi #1032: Odroid C2 | Debian Stretch beta image

Fourdee, 2017-06-25:
Using Meveric's image as base:
- https://forum.odroid.com/viewtopic.php?f=138&t=27449

Fourdee, 2017-06-28:
Completed.

Fourdee, 2017-06-29:
Reopened for testing.

Fourdee, 2017-06-29:
YMPD:
```
root@DietPi:~# systemctl stauts ympd -l
Unknown operation stauts.
```

Fourdee, 2017-06-29:
🈴 Dmesg:
```
[  704.363292] systemd[1]: apt-daily-upgrade.timer: Adding 46min 52.016859s random time.
[  704.413792] systemd[1]: syscall 279
[  704.413807] Code: aa0503e4 aa0603e5 aa0703e6 d4000001 (b13ffc1f)
[  704.413814] CPU: 2 PID: 1 Comm: systemd Not tainted 3.14.79+ #1
[  704.413820] task: ffffffc05b46c000 ti: ffffffc05b4e0000 task.ti: ffffffc05b4e0000
[  704.413831] PC is at 0x7faefe57b4
[  704.413835] LR is at 0x5564bd1d90
[  704.413839] pc : [<0000007faefe57b4>] lr : [<0000005564bd1d90>] pstate: 60000000
[  704.413843] sp : 0000007ff92de7f0
[  704.413846] x29: 0000007ff92de7f0 x28: 0000000000000002
[  704.413853] x27: 0000005564c63000 x26: 0000005564bf9000
[  704.413858] x25: 0000007ff92de9c8 x24: 0000000000000000
[  704.413863] x23: 0000005564bf9268 x22: 0000000000000002
[  704.413867] x21: 0000005564c64990 x20: 0000005564c61000
[  704.413872] x19: 0000007ff92de860 x18: 0000000000000000
[  704.413877] x17: 0000007faefe5790 x16: 0000005564c61528
[  704.413882] x15: 3a216e7e07020f19 x14: ff00000000000000
[  704.413887] x13: 0000000000000000 x12: 0000000000000003
[  704.413891] x11: 0000000000000003 x10: 0101010101010101
[  704.413896] x9 : 7f7f7f7f7f7f7f7f x8 : 0000000000000117
[  704.413901] x7 : 0000005564c82ee0 x6 : 0000005564c82ee0
[  704.413906] x5 : 0000005564c82ee0 x4 : 0000005564c9e388
[  704.413910] x3 : 0000007ff92de778 x2 : 820aa241e97bab98
[  704.413915] x1 : 0000000000000001 x0 : 0000005564c0a498
```

Fourdee, 2017-06-29:
PMS:
```
W: The repository 'http://dev2day.de/pms jessie InRelease' is not signed.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
 [Info] Apt-get is processing, please wait...

WARNING: The following packages cannot be authenticated!
  plexmediaserver-installer:armhf
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
 [Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken packages. DietPi-Software will now exit.
```

Fourdee, 2017-06-30:
MariaDB:
> https://www.debian.org/News/2017/20170617
In Stretch, the default MySQL variant is now MariaDB. The replacement of packages for MySQL 5.5 or 5.6 by the MariaDB 10.1 variant will happen automatically upon upgrade.

This renders https://github.com/Fourdee/DietPi/issues/894 null

Fourdee, 2017-06-30:
Xserver:
Required driver packages, not available in Meverics Stretch repo:
```
E: Unable to locate package aml-libs-odroid
E: Unable to locate package xf86-video-mali-odroid
```

MichaIng, 2017-10-07:
What is actually the benefit of Meveric's image as base for Odroid C2 image, compared to default Debian? If I understand right, Meveric basically tries the same than DietPi, purging Debian down to a minimal image. But DietPi already has a working process for all it's other images (besides RPi) and wouldn't it be easier to use the same base image for all devices (besides RPi, as Raspbian provides official supported kernel, driver packages and adjustments for RPi)?

Or does Meveric's Repo provide some other improvements, that are not available on default Debian Repo and outweigh the fact that Meveric's image is also still "Work In Progress" and seem to bring some additional difficulties/handling an additional base image for DiePi scripts?

Fourdee, 2017-10-07:
@MichaIng 
With DietPi, we try to use official images (stability), then optimize and configure for DietPi.

We use Meveric's images for Odroid because of the following:
- He provides excellent Debian minimal images, ideal for DietPi 
- He provides a repo, and maintains packages such as kernel updates, Kodi
- He is approachable, puts lots of his own time into his images and we want to promote that.

MichaIng, 2017-10-09:
Okay sounds reasonable :). Just asked since this seems to create some additional effort.

LexiconCode, 2017-12-24:
is this the most up-to-date dietpi image for C2 stretch?

Fourdee, 2017-12-25:
@LexiconCode 

> is this the most up-to-date dietpi image for C2 stretch?

Yes currently, automatically updates to v159 during 1st run. 
I've been running it for a few weeks. Aside from Kodi and GPU (Jessie packages only), it runs a dream.

MichaIng, 2018-01-09:
@Fourdee 
Regarding YMPD we can maybe just use our own libssl1.0.0 binary, as already done in several other cases? http://dietpi.com/downloads/binaries/all/libssl1.0.0_1.0.1t-1+deb8u7_arm64.deb
The issue should occur on all Stretch machines, not just Odroid C2.

Regarding Xserver and Kodi I guess we have to wait until official Meveric Stretch release.

Fourdee, 2018-01-28:
Marking as closed. With v6.0 we only offer Jessie images for Odroids. Pending Meveric's repo update to Stretch, we can then move our image forward to match.

closed 2018-01-28 by Fourdee