MichaIng/DietPi #1034: DietPi-Software | Optimize Nextcloud

sootstone, 2017-06-27:
Just thinking it might be nice to add some tweaks to DietPi's Nextcloud installation.

Namely:

1. Using Redis for [Transactional File Locking](https://docs.nextcloud.com/server/12/admin_manual/configuration_files/files_locking_transactional.html#transactional-file-locking) to avoid file corruption.

2. Using Cron to run background jobs. Default is AJAX (less reliable) - [https://docs.nextcloud.com/server/12/admin_manual/configuration_server/background_jobs_configuration.html#defining-background-jobs](https://docs.nextcloud.com/server/12/admin_manual/configuration_server/background_jobs_configuration.html#defining-background-jobs)

Just a few ideas, please add more :)

Yury-MonZon, 2017-07-08:
SSL encryption by default(https) would be great, especially if a user doesn't have domain name for his server.

MichaIng, 2017-07-18:
Sorry, had the same idea but was too blind to see this existing issue 🤕 . Just see/participate the issue above, where the topics here are handled as well.

Fourdee, 2017-07-19:
@MichaIng 

👍 

I'll mark this ticket as closed. Please continue discussions in the following ticket:
https://github.com/Fourdee/DietPi/issues/1067

closed 2017-07-19 by Fourdee