MichaIng/DietPi #1036: chroot to RPi-armv6-(Stretch) sd card fails: qemu: uncaught target signal 4 (Illegal instruction)

xanview, 2017-06-28:
All of what I try below works with the TinkerBoard image (DietPi_v149_AsusTinkerBoard-armv7-(Jessie).img), so I'm not sure what is different in the RasberryPi image. From googling, it seems this used to be a problem with old versions of Raspbian (Jessie), so not sure what is wrong:

I created a microSD card from the latest stretch image (v151):
```
sudo dd if=DietPi_v151_RPi-armv6-\(Stretch\).img of=/dev/rdisk6 bs=1m
```

I am trying to chroot to it from my Ubuntu machine (I have qemu-user-static installed):

```
$ mount /dev/sdb2 /mnt/target
$ mount /dev/sdb1 /mnt/target/boot
$ mount -t proc proc /mnt/target/proc
$ mount -t sysfs sysfs /mnt/target/sysfs
$ mount --bind /dev/ /mnt/target/dev
$ mount --bind /dev/pts /mnt/target/dev/pts
$ chroot /mnt/target
chroot: faield to run command '/bin/bash': No such file or directory
```

All the files are there, so that error doesn't seem very descriptive:

```
# ls -al /mnt/target/bin/cat /mnt/target/bin/grep /mnt/target/bin/bash
-rwxr-xr-x 1 root root 912712 Jun 12 18:24 /mnt/target/bin/bash
-rwxr-xr-x 1 root root  30560 Jun 12 18:24 /mnt/target/bin/cat
-rwxr-xr-x 1 root root 173656 Jun 12 18:24 /mnt/target/bin/grep
```

I also tried to explicitly use qemu:

```
# chroot /mnt/target qemu-arm-static /bin/bash
bash: /bin/cat: No such file or directory
bash: /bin/grep: No such file or directory
declare -x HOME="/root"
declare -x LANG="en_GB.UTF-8"
declare -x LANGUAGE="en_GB:en"
declare -x LESSCLOSE="/usr/bin/lesspipe %s %s"
declare -x LESSOPEN="| /usr/bin/lesspipe %s"
declare -x LOGNAME="root"
declare -x LS_COLORS="rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:"
declare -x MAIL="/var/mail/root"
declare -x OLDPWD
declare -x PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin"
declare -x PWD="/"
declare -x SHELL="/bin/bash"
declare -x SHLVL="2"
declare -x TERM="xterm-256color"
declare -x USER="root"
bash: /DietPi/dietpi/login: No such file or directory
qemu: uncaught target signal 4 (Illegal instruction) - core dumped
Illegal instruction (core dumped)
```

Any ideas? - I am able to chroot to the TinkerBoard DietPi image the same way as above.

MichaIng, 2017-11-27:
> bash: /DietPi/dietpi/login: No such file or directory

I am not too deep inside the technical details, but dietpi scripts assume some tmpfs mounts of `/boot/dietpi` to `/DietPi` e.g. to run, which are done during boot. No idea how/if this is handled on chroot.
Also on RPi the sdcard partitions have different /dev/ locations, usually `/dev/mmcblk0p1` for /boot and `/dev/mmcblk0p2` for root.
____________________
**About QEMU:**
I tried to start the DietPi-Stretch-RPi image on qemu-arm on windows, but failed. Needed to use special kernels for this: https://github.com/dhruvvyas90/qemu-rpi-kernel. It starts up first, but after initial first-run setup of DietPi it fails to mount the root partition and starts boot looping. Tried to fix it by manual script adjustments but failed. Gave up also a bid early, because qemu on windows without hardware VT support is extremely slow.

This guide here basically works. Some things can be adjusted, but others need to be exactly like stated there (kernel, ram size, cpu, machine): https://blog.agchapman.com/using-qemu-to-emulate-a-raspberry-pi/
As said, first bootup works, so should work with raspbian lite image, but DietPi first run setup breaks it for me. I guess we need to wait for some special qemu image, or build one by ourself, using raspbian lite as basement, as intended.

I mentioned qemu image already here: https://github.com/Fourdee/DietPi/issues/1219#issuecomment-340458923
But if this is really wanted, we should create a new issue for this.

MichaIng, 2018-01-31:
@xanview
Is this issue still current? Test the new DietPi v6.0 image in case, since we made heavy rework there.

At least I can say that I was able to chroot from RPi into a RPi DietPi image, plugged via USB stick.

Fourdee, 2018-02-22:
Closing due to no response.

closed 2018-02-22 by Fourdee