MichaIng/DietPi #1037: You cannot deinstall software if space is <500MB

Invictaz, 2017-06-29:
A bit of a vicious circle. You need to free up space, but you cannot do that since the script does not run when under 500MB. 

I understand the 500MB minimum requirement but now it's preventing me to free up space...

Fourdee, 2017-06-29:
@Invictaz 

> A bit of a vicious circle. 

Spot on 👍 

I'll switch the check around a bit, allow uninstalls but not installs.

Fourdee, 2017-06-29:
@Invictaz 

Completed, you can update the script now with:
```
wget https://github.com/Fourdee/DietPi/raw/c62ecb5ec68fb235421335dcdc49c4487d6d56f4/dietpi/dietpi-software -O /DietPi/dietpi/dietpi-software
```

Software uninstalls are now supported if free space is below 500MB threshold.

closed 2017-06-29 by Fourdee

Invictaz, 2017-06-29:
Legend!