MichaIng/DietPi #1038: NZBget not running

Invictaz, 2017-06-29:
Moved all userdata to my USB stick then installed NZBget

ps -A | grep nzbget 

shows nothing
Cannot connect to port 6789

Made a config file out of the template but it seems to be not running. In HTOP I see nothing.

Fourdee, 2017-06-29:
@Invictaz 

Lets check service status:
```
systemctl status nzbget.service -l
```

Invictaz, 2017-06-29:
I found what is wrong during package.run downloading 

[image]

Invictaz, 2017-06-29:
@Fourdee 

[image]

Invictaz, 2017-06-29:
Dietpi userdata is now on /mnt/USB_1/dietpi_userdata and not on /mnt/dietpi_userdata/

So a redirect is missing somewhere (also).

Invictaz, 2017-06-29:
I changed mount mode from /dev/UUID to /dev/SD in Drive Manager since FAT does not seem to support users. This did not work.

Fourdee, 2017-06-29:
@Invictaz 

FAT16/32 does not support file permissions. We do not recommend using it for user data.

You'll need to move your userdata back to RootFS. Then format USB drive with ext4. Then you can transfer user data back to USB drive.

Invictaz, 2017-06-29:
@Fourdee Thanks for the quick reply. Not exactly the solution I was hoping for but understandable.

Fourdee, 2017-06-30:
@Invictaz 

Yep, permissions will reduce filesystem performance, however, the majority of software requires them to function.

I'll mark this as closed. Please reopen if required.

closed 2017-06-30 by Fourdee