MichaIng/DietPi #1042: Image request for the Nano Pi K2

vsxen, 2017-06-30:
Amlogic S905 SoC  

Those links may be  helpsful.

http://freaktab.com/forum/tv-player-support/amlogic-based-tv-players/606133-linux-images-for-s802-s805-s812-s905-s905x-s912-sd-usb-emmc

https://yadi.sk/d/pHxaRAs-tZiei

http://www.mediafire.com/file/7re9051p3ptkn74/nanopi-k2%28armbian+patch%29.tar.gz

https://forum.armbian.com/index.php?/topic/4150-friendlyelec-nanopi-k2-s905/&

Thank you.

Fourdee, 2017-07-01:
@vsxen 

Our Odroid C2 image _may_ work with this (same ARM), although untested as I do not currently own the K2 board.

vsxen, 2017-07-01:
Ok,  I will try it .(May be  I will  need to compile u-boot  like the armbian patch file) Thank you.

Kinder0083, 2017-09-19:
Hello

Did you succeed to run DietPi on your Nanopi K2 board?
I have this card and want to install Dietpi.
How did you do ?

Thank you

matohavo, 2017-11-09:
+1
 @vsxen Did it work? Could you please explain how to use Odroid C2 image with K2 or at least provide some references? Thank you.

Kinder0083, 2017-11-09:
Hello. sorry it does not work.