MichaIng/DietPi #1048: DietPi-Config | LCD Panel addon: waveshare32 doesn't work in v.152 with RPi 3

k-plan, 2017-07-02:
## Creating a bug report/issue:

@Fourdee 

#### Required Information:
- DietPi Version |  => 152

- SBC Device : => RPi 3

```
- RPi Board Info -
 Revision         |     a02082
 Released         |     Q1 2016
 Model            |     3 Model B
 Memory           |     1024 MB
 Manufacturer     |     Sony
```

- Power supply used : => offical RPi Power Supply 5,1 V / 2.5 A

- SD card used : => ScanDisk Extreme 16GB U3

- Distro (EG: Jessie) :  ==> Linux RPI-3 4.9.30-v7+ #2 SMP Wed Jun 7 18:57:30 BST 2017 armv7l GNU/Linux

#### Additional Information (if applicable):

- Software title? => `dietpi-cloudshell` &  `LCD Panel addon: waveshare32`

- Can this issue be replicated on a fresh installation of DietPi? => Yes

#### Expected behaviour:

-  LCD Panel addon: waveshare32 should display dietpi-cloudshell

#### Actual behaviour:

- dietpi-cloudshell working on terminal or HDMI, but LCD Panel will only be white

[image]

#### Steps to reproduce:

- install via `dietpi-software` and `dietpi-config`

#### Did you submit a ```dietpi-bugreport```?

NO!

#### Extra details:

have a look here: [RPI3-waveshare32-white screen of brilliance-Diepti V152](http://dietpi.com/phpbb/viewtopic.php?f=11&t=2007)

and

```
root@RPI-3:~# dmesg
. . .
[    0.661396] vc-sm: Videocore shared memory driver
[    0.662495] [vc_sm_connected_init]: start
[    0.663781] vc_vchi_sm_init: failed to open VCHI service (-1)
[    0.664362] [vc_sm_connected_init]: failed to initialize shared memory service
[    0.665986] [vc_sm_connected_init]: end - returning -1
. . .
[    2.507975] Driver for 1-wire Dallas network protocol.
[    2.528518] w1-gpio 4.onewire: gpio pin 4, external pullup pin 1, parasitic power 0
[    2.528557] w1_add_master_device: set_pullup requires write_byte or touch_bit, disabling
[    2.628266] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
[    2.635304] spi-bcm2835 3f204000.spi: chipselect 1 already in use
[    2.635318] spi_master spi0: spi_device register error /soc/spi@7e204000/waveshare32b_ts@1
[    2.635331] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/waveshare32b_ts@1
[... 139 lines cut ...]
fbtft                  34966  1 fb_ili9340
joydev                  9988  0
syscopyarea             3225  1 fbtft
sysfillrect             3826  1 fbtft
ads7846                13656  0
brcmutil                9092  1 brcmfmac
sysimgblt               2480  1 fbtft
fb_sys_fops             1741  1 fbtft
hwmon                  10552  1 ads7846
cfg80211              542989  1 brcmfmac
rfkill                 20832  1 cfg80211
spi_bcm2835             7596  0
bcm2835_gpiomem         3940  0
w1_gpio                 4818  0
wire                   32619  1 w1_gpio
cn                      5889  1 wire
uio_pdrv_genirq         3923  0
uio                    10268  1 uio_pdrv_genirq
fixed                   3285  0
fuse                   99603  1
ipv6                  408530  24
```

cu
k-plan

Fourdee, 2017-07-02:
@k-plan 

Thanks for report (very nice report 😄 ) 👍 

Should be able to get round to this on Tue. Just finishing up: https://github.com/Fourdee/DietPi/issues/1007

k-plan, 2017-07-02:
@Fourdee 

> Thanks for report (very nice report 😄 ) 👍

No problem, you created a nice template. 😉 

With older Kernel `Linux DietPi 4.4.50+ #1 Tue Mar 7 15:10:58 UTC 2017 armv6l GNU/Linux` e.g. DietPi v.14x it will work _"out-of-the-box"_.

Do some deeper testing and research on a fresh installed `DietPi_v150_RPi-armv6-(Jessie).img`

It exist a new Driver packet/ Online-Installer: http://www.waveshare.com/wiki/3.2inch_RPi_LCD_(B)#Driver

and some hints: (http://www.waveshare.com/wiki/3.2inch_RPi_LCD_(B)#Method_1._Driver_installation)

> Note: this LCD won't work after apt-get upgrade, in such cases, please edit the config.txt file in the SD card and remove this statement: dtoverlay=ads7846

and

How LCD to work with a camera at same time (picamera): http://www.waveshare.com/wiki/3.2inch_RPi_LCD_(B)#How_to_work_with_a_camera

Result: (only tested with `dietpi-cloudshell`, not tested with Desktop and Touchscreen!)

[image]

```
root@RPi-3_no-WiFi:~# cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11
```

```
root@RPi-3_no-WiFi:~# cat /DietPi/config.txt
. . .
#Waveshare 32 LCD
dtoverlay=waveshare32b:rotate=270
## dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,swapxy=0,pmax=255,xohms=60,xmin=200,xmax=3900,ymin=200,ymax=3900
dtoverlay=w1-gpio-pullup,gpiopin=4,extpullup=1
. . .
```

```
root@RPi-3_no-WiFi:~# ls -lah /boot/overlays | grep wave
-rwxr-xr-x 1 root root 2.1K Jul  2 14:21 waveshare32b.dtbo
-rwxr-xr-x 1 root root 2.1K Jul  2 14:21 waveshare32b-overlay.dtb

```

```
> root@RPi-3_no-WiFi:~# ls -lah LCD-show
> total 144K
> drwxrwxrwx 6 root   root    4.0K Mar  8 14:13 .
> drwxr-xr-x 6 root   root    4.0K Jul  2 14:17 ..
> drwxrwxrwx 2 root   root    4.0K Jul  9  2016 ?
> drwxrwxrwx 2 root   root    4.0K Mar  8 13:12 boot
> -rwxrwxrwx 1 root   root     168 Jul  9  2016 cmdline.txt
> -rwxrwxrwx 1 root   root    1.6K Mar 16  2016 dtc.sh
> drwxrwxrwx 3 root   root    4.0K Aug  4  2016 etc
> -rwxrwxrwx 1 root   root    2.1K Jul  9  2016 inittab
> -rwxrwxrwx 1 root   root    1.7K Mar  8 17:02 LCD101-1024x600-show
> -rwxr-xr-x 1 dietpi dietpi  1.7K Mar  8 17:39 LCD28-show
> -rwxrwxrwx 1 nobody nogroup 1.7K Mar  8 17:42 LCD32-show
> -rwxr-xr-x 1 root   root    1.7K Mar  8 17:35 LCD35B-show
> -rwxr-xr-x 1 dietpi dietpi  1.7K Mar  8 17:37 LCD35CPLD-show
> -rwxrwxrwx 1 root   root    1.7K Mar  8 17:21 LCD35-show
> -rwxrwxrwx 1 root   root    1.6K Mar  8 16:05 LCD43-show
> -rwxr-xr-x 1 root   root    1.6K Mar  8 17:44 LCD4-800x480-show
> -rwxrwxrwx 1 root   root    1.6K Mar  8 17:45 LCD4-show
> -rwxrwxrwx 1 root   root    1.6K Mar  8 17:47 LCD5-show
> -rwxrwxrwx 1 root   root    1.6K Mar  8 18:45 LCD7-1024x600-show
> -rwxrwxrwx 1 root   root    1.6K Mar  8 17:53 LCD7-800x480-show
> -rwxrwxrwx 1 root   root     264 Jul  9  2016 LCD-hdmi
> drwxrwxrwx 3 root   root    4.0K Jul  9  2016 usr
> -rwxrwxrwx 1 root   root    2.1K Jul  9  2016 waveshare32b-overlay.dtb
> -rwxrwxrwx 1 root   root    2.7K Jul 28  2016 waveshare35a-overlay.dtb
> -rwxr-xr-x 1 root   root    2.6K Oct 10  2016 waveshare35b-overlay.dtb
> -rwxr-xr-x 1 dietpi dietpi  2.7K Feb 21 04:22 waveshare35cpld-overlay.dtb
> -rwxr--r-- 1 nobody nogroup  38K Jul  9  2016 xinput-calibrator_0.7.5-1_armhf.deb
```

Tested with WiFI and/or ethernet connection - works.

k-plan, 2017-07-02:
@Fourdee 

can't get X11 working on RPi fbdev (LXE Desktop): https://github.com/Fourdee/DietPi/issues/409#issuecomment-232729020

But I still think, use it for DietPi Cloudshell on a 3.2'' display , its more than enough.

Fourdee, 2017-07-05:
@k-plan 

Bumped to v154. Need to get v153 wrapped up.

Fourdee, 2017-07-12:
Notes:
- The new driver installer is really messy. Requires a custom build of device tree compiler to install.

Options:
- Install 4.4 kernel along side the existing driver?
- Attempt to make the new driver "somewhat functional and simple"? Backtracking all the changes in the driver.

Fourdee, 2017-07-14:
- 🈴 Still not functional under X server. Tried both ```fbturbo``` and ```fbdev``` drivers. No joy.

Fourdee, 2017-07-14:
Resolved for v154. @k-plan Thanks for the update driver, latest DTB overlay did the trick 👍

closed 2017-07-14 by Fourdee