MichaIng/DietPi #1049: DietPi System | Option to set a custom NTP Server address

k-plan, 2017-07-02:
@Fourdee 

It will be a handy and nice feature, if we can set a custom NTP Sever address (IP or fully qualified domain name) directly in `dietpi-config` and/or `dietpi.txt`.

For new users, a listing of some predefine NTP  Server,  how as well in `Choose Apt Mirror` would be very helpful.

Thanks in advance for your efforts.

cu
k-plan

Fourdee, 2017-07-02:
@k-plan 

Yep, good suggestion 👍  I'll take a look for v154.

Fourdee, 2017-07-11:
> http://www.pool.ntp.org/zone/europe
In most cases it's best to use pool.ntp.org to find an NTP server (or 0.pool.ntp.org, 1.pool.ntp.org, etc if you need multiple server names). The system will try finding the closest available servers for you. If you distribute software or equipment that uses NTP, please see our information for vendors.

Fourdee, 2017-07-11:
[image]
[image]

Fourdee, 2017-07-11:
Also added to ```dietpi.txt```, applied during 1st run automation.

Fourdee, 2017-07-12:
Completed.

closed 2017-07-12 by Fourdee