MichaIng/DietPi #1050: DietPi-Software | Searx Meta Search Engine

DJCrashdummy, 2017-07-03:
i know with [YaCy](https://github.com/Fourdee/DietPi/issues/817) there is already a search engine available, but [Searx](https://github.com/asciimoo/searx) is not completely the same:
it is a meta search engine and relies on the indexes from other providers (simplified you can think about it like your own free DuckDuckGo), but therefore it is not so heavy like YaCy.

as an lightweight alternative to YaCy it would be nice to have Searx as an option.

there is also some documentation for installing: https://asciimoo.github.io/searx/dev/install/installation.html

kaffeekanne, 2017-11-05:
As public searx instances are often very slow due to heavy traffic it would be awesome to have this officially included in dietpi. Will probably install it by hand, but for the interested non expert an automated installation would be nice.

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

b1n4ryj4n, 2019-11-08:
still not implemented?

Nutjob, 2019-11-10:
I would also like an implementation of Searx on Dietpi

IAmCainey, 2021-09-11:
I have been using Searx for a little bit now and its the only way i browse the net,. Cmon :)

surtarso, 2022-08-07:
kudos for searx!!  yacy , tho awesome, is too intense for simple use

tmcgough, 2022-12-13:
Seems all avenues have been closed, but I would like to add my vote for searx or searxng as both are very useful and would love to see them added to the list of tools on DietPi.  YaCy can fill up your storage if left unchecked and not very verbose on searrches.  Searx and Searxng are both very efficient and work great on the DietPi.  As you state, they can be installed manually but it would be very nice to see them as an option along with YaCy in the approved added installation options.