MichaIng/DietPi #1056: DietPi 153 broke Pimatic and Homeduino

yaba, 2017-07-05:
Yesterday I've updated from 148 to 153 (RPi 2).
Is is possible to downgrade? Pimatic uses homeduino which in turn uses a process that is now always using CPU at 90% (and more).

Linux PiLight 4.9.35+ #1 Tue Jul 4 17:16:26 UTC 2017 armv6l GNU/Linux

Invictaz, 2017-07-05:
@yaba 

I believe if you download this tree: 

https://github.com/Fourdee/DietPi/tree/4e108f958e2e8f5c49ddc4a42ac22d9417f28719

You have back v152 again. 

Don't know for sure if that is the original download procedure as on dietpi.com/download older versions are not stored, only the images.

@Fourdee describes the upgrade process as:

> Its a basic file pull/download of https://github.com/Fourdee/DietPi/blob/master/dietpi/server_version. This contains current available version of DietPi.
> Updates are checked during each boot of system, and, when user manually runs dietpi-update.
> 
> If update is available and user selects to update, a Zip of the git master branch is pulled, unzipped and replaces current DietPi scripts:
> https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-update
> 
> Patches are executed incrementally during update process:
> 
> loop: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-update#L219-L238
> patch file: https://github.com/Fourdee/DietPi/blob/master/dietpi/patch_file

Maybe if you change the server version file to 152 it downloads 152 again? Not sure.

Fourdee, 2017-07-06:
@yaba 

> Pimatic and Homeduino

Other than the recent kernel update to ```4.9.35```, there is nothing in the update that targets those programs in any way.

Can you link the installation steps/guide you used for these pieces of software? Hopefully we can backtrack the install/configurations, that may be effected (if at all) from the update.

yaba, 2017-07-09:
@Fourdee 

Yes, vhduino is not compatible with the new kernel due to some issues in WiringPi.
Please close, not related to DietPi directly.

Fourdee, 2017-07-10:
@yaba 

> Yes, vhduino is not compatible with the new kernel due to some issues in WiringPi.
Please close, not related to DietPi directly.

Thanks for the info 👍 

WiringPi hasn't been updated since ```2017-03-03``` (https://git.drogon.net/?p=wiringPi;a=summary), so if that is the issue, try reverting to 4.4 kernel:
```
wget http://dietpi.com/downloads/binaries/rpi/dietpi_rpi_kernel.zip
unzip -o dietpi_rpi_kernel.zip -d /
reboot
```

Fourdee, 2017-07-10:
@yaba 
I'll mark as closed, but if you get a chance to test 4.4 kernel (https://github.com/Fourdee/DietPi/issues/1056#issuecomment-314079991), please let us know if this resolves the issue.

closed 2017-07-10 by Fourdee