MichaIng/DietPi #1058: No settings for composite video output

Invictaz, 2017-07-06:
In resolution setting there is no such thing as 720x576 (PAL)

Also there is only an overscan setting yes or no but no overscan compensation values, see here

[image]

Fourdee, 2017-07-13:
```
sdtv_mode=0    Normal NTSC
 sdtv_mode=1    Japanese version of NTSC – no pedestal
 sdtv_mode=2    Normal PAL
 sdtv_mode=3    Brazilian version of PAL – 525/60 rather than 625/50, different subcarrier
```

Fourdee, 2017-07-13:
Overscan Compensation becomes available once Overscan is enabled:
[image]

Fourdee, 2017-07-13:
- 🈯️ Test overscan
- 🈺 Test SDTV: Unable, lack composite device.

Fourdee, 2017-07-13:
@Invictaz 

Any chance you would be willing to test the SDTV modes in testing branch for us?
https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md

Fourdee, 2017-07-16:
I'll mark this as completed. Testing outstanding, however, confident functional.

@Invictaz 
If you get a chance to test this, please let us know.

closed 2017-07-16 by Fourdee

Invictaz, 2017-07-16:
@Fourdee How to update from testing branch?