MichaIng/DietPi #1059: Failure to Install Emby Server

DncrJim, 2017-07-07:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 153
- SBC Device | RPi 3

#### Additional Information (if applicable):
Fresh install of DietPi fails to install emby-server using 'dietpi-software'.

"http://ftp.us.debian.org/debian/pool/main/libj/libjpeg8/libjpeg8_8d1-2_amd64.deb is offline and/or unreachable."

Perhaps the solution is obvious and/or simple, but I have exhausted my skill level and forum/google search to no avail. 

I have not submitted a bug report because the system auto-restarts after failure to install. If system info etc. is necessary for debug, I will be happy to submit the report.

Note: a new version of emby-server was released early this week (and failing to auto-upgrade, prompting this ill-conceived and apparently futile attempt at a fresh install); I do not know if this is relevant.

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2017-07-07:
@DncrJim 

Thanks for report. Confirmed.

Seems the connection test fails:
```
[Info] Testing connection to http://ftp.us.debian.org/debian/pool/main/libj/libjpeg8/libjpeg8_8d1-2_amd64.deb
 [Info] Max duration of 20 seconds, please wait...
 [Failed] Connection test | An issue has occured
```

Fourdee, 2017-07-07:
libjpeg8 ARM64 available only in SID:
https://packages.debian.org/sid/libjpeg8

Only other available package is in Wheezy:
https://packages.debian.org/wheezy/libjpeg8

Fourdee, 2017-07-07:
@DncrJim 

Fixed for v154.

You can patch the script now with:
```
wget https://github.com/Fourdee/DietPi/raw/testing/dietpi/dietpi-software -O /DietPi/dietpi/dietpi-software
```

Emby server will now install in ```dietpi-software```.

closed 2017-07-07 by Fourdee