MichaIng/DietPi #1063: Banana Pi M3 official support

stdusm, 2017-07-08:
This is a request to  support Banana Pi M3 officially.
I use it for more than one year without any issues with latest updates and so on.
Here is the sample:
http://forum.banana-pi.org/t/bpi-m3-new-image-dietpi-preview-bpi-m3-sd-emmc-image-2016-5-14/1685 
Thanks.

Fourdee, 2017-07-08:
@stdusm 

> http://forum.banana-pi.org/t/bpi-m3-new-image-dietpi-preview-bpi-m3-sd-emmc-image-2016-5-14/1685

This is an unofficial image created by BPi, not provided/supported by us as we lack the board currently.

I'll pop the BPi M3 on the SBC purchase list.

Elluminatus, 2017-08-17:
Would be very nice because of the sata port and the cpu performance. Nice piece of hardware for a little server in the house.

zgon81, 2017-12-15:
What is the status? It would be really great to officialy implement the DietPi on my BananaPi M3...

Fourdee, 2017-12-16:
@zgon81 

> What is the status? 

Unfortunately, still lacking the board for implementation. Price of unit is also in question, beyond current funds.

Fourdee, 2018-02-22:
Closing due to BPi image/support being dropped.

closed 2018-02-22 by Fourdee