MichaIng/DietPi #1065: Google Assistant SDK and AIY kit.

R1CH-T, 2017-07-09:
Hey Dan,

A while back, I received one of those funky, DIY, cardboard Google assistants from the MagPi:
[https://aiyprojects.withgoogle.com/voice/](url)

I decided to have a go at adding this functionality to DietPi, by adding two new installation options to DietPi-Software: One for the standard Assistant SDK and one for the AIY kit.

For each of these options, I have tried to implement the instructions at each of these sites, respectively:

SDK: [https://developers.google.com/assistant/sdk/prototype/getting-started-pi-python/run-sample](url)

AIY Kit: [https://github.com/google/aiyprojects-raspbian/blob/master/HACKING.md#setting-up-the-image](url)

Could you please take a look at my attempt at this in the link below and let me know if I have done it correctly. I will have a go at testing this soon, but I wanted you to take a look firs; this is my first attempt at doing this sort of thing myself.

[https://github.com/R1CH-T/DietPi/commit/b5b8f1cdf6be76be374d1907161e6e6218757248](url)

Thanks, Rich.

R1CH-T, 2017-07-09:
I haven't tested the SDK installer yet, but I've gotten the installer for the AIY kit to work, by adding these to dietpi-software:

Added to the end of the:
> aSOFTWARE_CATEGORIES_DIETPI=(

`"────Voice Control─────────────────────────────────────────" #20`

Just after the section:

>  #Printing

```
		#Voice Control
		#--------------------------------------------------------------------------------
		index_current=154

				 aSOFTWARE_WHIP_NAME[$index_current]='Google Assistant SDK'
				 aSOFTWARE_WHIP_DESC[$index_current]='Google Assistant SDK for any device'
			aSOFTWARE_CATEGORY_INDEX[$index_current]=20
					  aSOFTWARE_TYPE[$index_current]=0
			 aSOFTWARE_ONLINEDOC_URL[$index_current]='???????'
				aSOFTWARE_REQUIRES_ALSA[$index_current]=1
					
		#------------------
		index_current=155

				 aSOFTWARE_WHIP_NAME[$index_current]='Google Assistant AIY Kit'
				 aSOFTWARE_WHIP_DESC[$index_current]='Google AIY Kit for Raspberry Pi'
			aSOFTWARE_CATEGORY_INDEX[$index_current]=20
					  aSOFTWARE_TYPE[$index_current]=0
			 aSOFTWARE_ONLINEDOC_URL[$index_current]='???????'
				aSOFTWARE_REQUIRES_ALSA[$index_current]=1
		aSOFTWARE_REQUIRES_BUILDESSENTIAL[$index_current]=1
				aSOFTWARE_REQUIRES_GIT[$index_current]=1
		
		# - Disabled for All non-rpi
		for ((i=10; i<=$MAX_HW_MODEL; i++))
		do

			aSOFTWARE_AVAIL_HW_MODEL[$index_current,$i]=0

		done
		
		#------------------
```

Just before:
> 	Install_Linux_Software(){

	```
	#GoogleVoiceSDK
		INSTALLING_INDEX=154
		if (( ${aSOFTWARE_INSTALL_STATE[$INSTALLING_INDEX]} == 1 )); then

			Banner_Installing
			
			AGI python3-dev python3-venv
			
			python3 -m venv env
			env/bin/python -m pip install --upgrade pip setuptools
			source env/bin/activate
			
			python -m pip install --upgrade google-assistant-library
			
			python -m pip install --upgrade google-auth-oauthlib[tool]
						
			INSTALL_URL_ADDRESS='https://www.googleapis.com/auth/assistant-sdk-prototype'
			/DietPi/dietpi/func/check_connection "$INSTALL_URL_ADDRESS"

			if (( $? == 0 )); then
			
			cp /boot/client_secret_*.json "$FP_DIETPI_USERDATA_DIRECTORY"/assistant.json
			
			google-oauthlib-tool --client-secrets "$FP_DIETPI_USERDATA_DIRECTORY"/assistant.json --scope "$INSTALL_URL_ADDRESS" --save --headless
			
			
			
			else

				Error_NoConnection_NoInstall

			fi

		fi
		
		#GoogleVoiceAIY
		INSTALLING_INDEX=155
		if (( ${aSOFTWARE_INSTALL_STATE[$INSTALLING_INDEX]} == 1 )); then

			Banner_Installing

			INSTALL_URL_ADDRESS='https://github.com/google/aiyprojects-raspbian.git'
			/DietPi/dietpi/func/check_connection "$INSTALL_URL_ADDRESS"

			if (( $? == 0 )); then

				AGI python python-dev

				git clone "$INSTALL_URL_ADDRESS" "$FP_DIETPI_USERDATA_DIRECTORY"/voice-recognizer-raspi

				cd "$FP_DIETPI_USERDATA_DIRECTORY"/voice-recognizer-raspi
				
				scripts/install-deps.sh
				scripts/install-services.sh
				
				AGI install raspberrypi-kernel
				scripts/configure-driver.sh
				scripts/install-alsa-config.sh
				
				cp /boot/client_secret_*.json "$FP_DIETPI_USERDATA_DIRECTORY"/assistant.json
				
				src/main.py

			else

				Error_NoConnection_NoInstall

			fi

		fi
```

It gets all the way through the installation, but when I try to run the main.py command I just get exceptions from Python. Examples:

```
root@AIY:/mnt/dietpi_userdata/voice-recognizer-raspi# python src/main.py
Traceback (most recent call last):
  File "src/main.py", line 26, in <module>
    import configargparse
ImportError: No module named configargparse
```

```
root@AIY:/mnt/dietpi_userdata/voice-recognizer-raspi# python3 src/main.py
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/google/auth/transport/grpc.py", line 20, in <module>
    import grpc
ImportError: No module named 'grpc'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "src/main.py", line 32, in <module>
    import speech
  File "/mnt/dietpi_userdata/voice-recognizer-raspi/src/speech.py", line 26, in <module>
    import google.auth.transport.grpc
  File "/usr/local/lib/python3.4/dist-packages/google/auth/transport/grpc.py", line 23, in <module>
    'gRPC is not installed, please install the grpcio package to use the '
ImportError: gRPC is not installed, please install the grpcio package to use the gRPC transport.

```
Very frustrating!

R1CH-T, 2017-07-10:
Interesting link: [](url)http://eduncan911.com/stem/embedded/aiy-google-assistant-upgrades-may-2017.html

R1CH-T, 2017-07-10:
I have opened an issue on Google's AIO projects Github:

https://github.com/google/aiyprojects-raspbian/issues/94

Fourdee, 2017-07-10:
@R1CH-T 

Hi Rich,

> P.S. Do you have one of the AIY Kits, yet?

I don't, but I believe i'll need to obtain one if I'am to be any assistance here. It seems a large implementation that would need much testing, although you've done great so far, looks good.

> ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
arecord: main:722: audio open error: No such file or directory
[2017-07-10 00:45:43,804] ERROR:audio:Microphone recorder died unexpectedly, aborting...

Seems the microphone is failing open a ALSA stream. If the AIY kit image works flawlessly, it could be a lack of documentation for the manual installation, or a slip in the configuration somewhere.

> Could you please take a look at my attempt at this in the link below and let me know if I have done it correctly. 

It looks all in order, however, I usually have everything outside of any DietPi scripts, in note form, before I implement it.

You will have issues with your ```/etc/asound.conf```, this will be overridden during ```dietpi-software``` installation at some point, i'll need to add support for this special configuration.

---

Cant see to find a copy of the MagPi 57 anywhere for purchase lol. I'am assuming the voice hat is required for this project (https://github.com/google/aiyprojects-raspbian)?

Fourdee, 2017-07-10:
@R1CH-T 

It may be worth trying https://github.com/google/aiyprojects-raspbian/blob/master/HACKING.md#setting-up-the-image on Raspbian. If that is successful, we could compare the installs and try to find out why it only fails on DietPi.

Fourdee, 2017-07-10:
@R1CH-T 
```
root@AIY:/mnt/dietpi_userdata/voice-recognizer-raspi# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
softvol
micboost
default
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
```

Doesnt seem like the dtoverlay is loaded for voice hat?

Are these applied to ```/DietPi/config.txt``` (our RAMdisk location for ```config.txt```:
https://github.com/google/aiyprojects-raspbian/blob/master/scripts/configure-driver.sh#L27-L35

R1CH-T, 2017-07-10:
Thanks for the feedback. :1st_place_medal:
I was up very late last night looking at this stuff, so I've just read these replies.

> It may be worth trying https://github.com/google/aiyprojects-raspbian/blob/master/HACKING.md#setting-up-the-image on Raspbian. If that is successful, we could compare the installs and try to find out why it only fails on DietPi.

I was thinking along those lines, last night, before my brain melted. ;)

> Doesnt seem like the dtoverlay is loaded for voice hat?
Are these applied to /DietPi/config.txt (our RAMdisk location for config.txt:
https://github.com/google/aiyprojects-raspbian/blob/master/scripts/configure-driver.sh#L27-L35
Applied to /boot/config.txt, settings will be lost during reboot. Needs to be applied to our RAMdisk location /DietPi/config.txt.

I'll take a look. I won't have as much time to hack on this during the week, but I'll do my best. :)

> Cant see to find a copy of the MagPi 57 anywhere for purchase lol. I'am assuming the voice hat is required for this project (https://github.com/google/aiyprojects-raspbian)?

Hehe. Yeah. I think that every copy on the planet sold within two days!
I'd send you mine, but it kinda defeats the object...
There is a wait-list though, if it helps: https://docs.google.com/forms/d/e/1FAIpQLSev7IQBFUaDlv5tx1Decxd5Ya5AqYSEvD72hJySeaRDogaqAw/viewform?c=0&w=1

[I've gotten a reply over at the other issue page](https://github.com/google/aiyprojects-raspbian/issues/94), so I'll have a go with his suggestion there, too:

>Could you run the checkpoint scripts:
https://github.com/google/aiyprojects-raspbian/tree/master/checkpoints
Also did use disable the HDMI audio? You should get this device listing:

>$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
softvol
micboost
default
sysdefault:CARD=sndrpigooglevoi
    snd_rpi_googlevoicehat_soundcar, 
    Default Audio Device
dmix:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar, 
    Direct sample mixing device
dsnoop:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar, 
    Direct sample snooping device
hw:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar, 
    Direct hardware device without any conversions
plughw:CARD=sndrpigooglevoi,DEV=0
    snd_rpi_googlevoicehat_soundcar, 
    Hardware device with all software conversions

>Another thing to check is whether the driver got loaded:

>$ lsmod | grep snd
snd_soc_bcm2835_i2s     6546  2 
snd_soc_googlevoicehat_soundcard     3351  3 
snd_soc_googlevoicehat_codec     3936  1 
snd_soc_core          179915  3 snd_soc_googlevoicehat_codec,snd_soc_googlevoicehat_soundcard,snd_soc_bcm2835_i2s
snd_compress           10384  1 snd_soc_core
snd_pcm_dmaengine       5894  1 snd_soc_core
snd_pcm                98501  4 snd_pcm_dmaengine,snd_soc_googlevoicehat_soundcard,snd_soc_core
snd_timer              23904  2 snd_pcm
snd                    70032  8 snd_compress,snd_timer,snd_soc_core,snd_pcm

It could be something to do with HDMI: I set it up headless with `/boot/dietpi.txt`, but didn't completely disable it.
It also does look like it's a problem with drivers not loading...

R1CH-T, 2017-07-10:
> Doesnt seem like the dtoverlay is loaded for voice hat?
>
> Are these applied to /DietPi/config.txt (our RAMdisk location for config.txt: [...]

Bingo!

R1CH-T, 2017-07-10:
https://github.com/google/aiyprojects-raspbian/tree/master/scripts

scripts/install-deps.sh:
```
...
for config in status-led.ini voice-recognizer.ini; do
  if [[ ! -f "${HOME}/.config/${config}" ]] ; then
    echo "Installing ${config}"
    cp "config/${config}.default" "${HOME}/.config/${config}"
  fi
```
scripts/install-services.sh:
```
...
for service in systemd/*.service; do
  sed "s:/home/pi/voice-recognizer-raspi:${repo_path}:g" "$service" \
    > "/lib/systemd/system/$(basename "$service")"
done
...
```

scripts/configure-driver.sh:
```
...
sed -i \
  -e "s/^dtparam=audio=on/#\0/" \
  -e "s/^#\(dtparam=i2s=on\)/\1/" \
  /boot/config.txt
grep -q "dtoverlay=i2s-mmap" /boot/config.txt || \
  echo "dtoverlay=i2s-mmap" >> /boot/config.txt
grep -q "dtoverlay=googlevoicehat-soundcard" /boot/config.txt || \
  echo "dtoverlay=googlevoicehat-soundcard" >> /boot/config.txt
grep -q "dtparam=i2s=on" /boot/config.txt || \
  echo "dtparam=i2s=on" >> /boot/config.txt
```
scripts/install-alsa-config.sh:
```
...
asoundrc=/home/pi/.asoundrc
global_asoundrc=/etc/asound.conf

for rcfile in "$asoundrc" "$global_asoundrc"; do
  if [[ -f "$rcfile" ]] ; then
    echo "Renaming $rcfile to $rcfile.bak..."
    sudo mv "$rcfile" "$rcfile.bak"
  fi
done
```

Fixing these should do the job, hopefully...

R1CH-T, 2017-07-10:
Would it be possible to use some sort of globally applied alias' to catch these sorts of things in dietpi? Or maybe some symlinks, or something, so we don't have to alter any such code again? Some catch-alls?

R1CH-T, 2017-07-10:
Success!

I just disabled the sound through Audio Options, changed the scripts and it works!

The light in the arcade button on top triggers voice recognition, doesn't flash like it should and the service is broken; but I think we can both see what's wrong, there:
```
root@AIY:~# systemctl status voice-recognizer.service
● voice-recognizer.service - voice recognizer
   Loaded: loaded (/lib/systemd/system/voice-recognizer.service; enabled)
   Active: failed (Result: start-limit) since Mon 2017-07-10 18:31:02 UTC; 1min 30s ago
  Process: 582 ExecStart=/home/pi/voice-recognizer-raspi/env/bin/python3 -u src/main.py (code=exited, status=217/USER)
```
scripts/install-services.sh:
```
set -o errexit

if [[ $EUID -ne 0 ]]; then
   echo "This script must be run as root (use sudo)" 1>&2
   exit 1
fi

cd "$(dirname "${BASH_SOURCE[0]}")/.."
repo_path="$PWD"

for service in systemd/*.service; do
  sed "s:/mnt/dietpi_userdata/voice-recognizer-raspi:${repo_path}:g" "$service" \
    > "/lib/systemd/system/$(basename "$service")"
done

# voice-recognizer is not enabled by default, as it doesn't work until
# credentials are set up, so we explicitly enable the other services.
systemctl enable alsa-init.service
systemctl enable ntpdate.service
systemctl enable status-led.service
systemctl enable status-led-on.service
systemctl enable status-led-off.service
systemctl enable status-monitor.service
```

scripts/configure-driver.sh:
```
set -o errexit

if [[ $EUID -ne 0 ]]; then
   echo "This script must be run as root (use sudo)" 1>&2
   exit 1
fi

set -e

sed -i \
  -e "s/^dtparam=audio=on/#\0/" \
  -e "s/^#\(dtparam=i2s=on\)/\1/" \
  /DietPi/config.txt
grep -q "dtoverlay=i2s-mmap" /DietPi/config.txt || \
  echo "dtoverlay=i2s-mmap" >> /DietPi/config.txt
grep -q "dtoverlay=googlevoicehat-soundcard" /DietPi/config.txt || \
  echo "dtoverlay=googlevoicehat-soundcard" >> /DietPi/config.txt
grep -q "dtparam=i2s=on" /DietPi/config.txt || \
  echo "dtparam=i2s=on" >> /DietPi/config.txt
```
scripts/install-alsa-config.sh:
```
set -o errexit

if [[ $EUID -ne 0 ]]; then
   echo "This script must be run as root (use sudo)" 1>&2
   exit 1
fi

cd "$(dirname "${BASH_SOURCE[0]}")/.."

asoundrc="${HOME}/.asoundrc"
global_asoundrc=/etc/asound.conf

for rcfile in "$asoundrc" "$global_asoundrc"; do
  if [[ -f "$rcfile" ]] ; then
    echo "Renaming $rcfile to $rcfile.bak..."
    sudo mv "$rcfile" "$rcfile.bak"
  fi
done

sudo cp scripts/asound.conf "$global_asoundrc"
echo "Installed voiceHAT ALSA config at $global_asoundrc"
```

Once those services in the first script are all sorted out, it should work perfectly!

Fourdee, 2017-07-11:
@R1CH-T 

> scripts/install-services.sh:
...
for service in systemd/*.service; do
  sed "s:/home/pi/voice-recognizer-raspi:${repo_path}:g" "$service" \
    > "/lib/systemd/system/$(basename "$service")"
done
...
(tries to use /home/pi)
If this is too problematic, perhaps the code could be altered upstream to make it a little friendlier to non-standard distros.
Changing where ${HOME} etc point to would be useful.

Yep, they could use ```"$HOME"/voice-recognizer-raspi``` to support the current user installing the software.

> scripts/configure-driver.sh:

I can make this easier by adding the soundcard as an option in ```dietpi-config```. Selecting it would have the same effect. Can also apply the asound.conf at the same time.

--- 

> /home/pi/voice-recognizer-raspi/env/bin/python3 -u src/main.py

Try running that command manually, if it fails, should give you some verbose information.

> It'd be handy to have stuff redirected to /mnt/dietpi_userdata in most cases

Yep 👍

R1CH-T, 2017-07-14:
I haven't had a lot of time to look at this over the last couple of days, but I have just been looking through 'dietpi-config'. I am trying to figure out how to add an option for the _AOI Hat_ sound hardware. I'm trying to get this down pat, which may take a little time, but I'd just like to ask if you have any templates or tools, which help you construct the statements to do this stuff.

I'd also like to add an option to install a _relay driver_ called [Crelay](https://github.com/ondrej1024/crelay), which allows the control of USB and GPIO relays from Linux machine via the command line, as well as a simple web interface. This would be very useful for hardware hackers, (including myself) and will expand the appeal of DietPi (especially if voice control can be integrated). ;)

R1CH-T, 2017-07-15:
I've been at work all week and  did the best I could, but I have finally had the chance to go through the `dietpi-config` code. Again; I am still at a loss: This script is 5000+ lines of code and I am having a little bit more trouble with this than I had with the `dietpi-installer`.

The `dietpi-installer` is pretty straight-forward, but I keep getting lost because of a combination of ignorance, inexperience and  'various distractions' (i.e. Family, etc.). So: Could you please give us some _read-me's_, as a follow-on from [https://github.com/Fourdee/DietPi/issues/490#issuecomment-244416570](https://github.com/Fourdee/DietPi/issues/490#issuecomment-244416570)?

Fourdee, 2017-07-15:
@R1CH-T 

Hi Rich,

> but dietpi-config seems more complex. So: Could you please give us some more tips, as a follow-on from #490 (comment)?

Yep, DietPi-Config uses multiple external script functions to apply its settings. The main one is ```dietpi-set_hardware```, which applies soundcards and such.

Is the DietPi-Config additions you are trying to apply, for the voice HAT soundcard? 

>  I keep getting lost because of a combination of ... 'various distractions' (i.e. Family, etc.).

Know the feeling lol 👍

Fourdee, 2017-07-15:
@R1CH-T 

> I'd also like to add an option to install a relay driver called Crelay, which allows the control of USB and GPIO relays from a Linux machine, via the command line, as well as providing a simple web interface. This would be very useful for hardware hackers (including myself), and will expand the appeal of DietPi (especially if voice control can be integrated). ;)

I'll take a look after the HAT steps are done :)

Fourdee, 2017-07-15:
@R1CH-T 

Guide for adding soundcards: 
https://github.com/Fourdee/DietPi/issues/1075#issuecomment-315530565

Should cover it, but let me know if any issues/unsure.

Fourdee, 2017-07-15:
@R1CH-T 

> https://github.com/ondrej1024/crelay EDIT: I have gotten this to work on a Linux x86-64 desktop and a Raspberry Pi (in minutes) by following the instructions on the GitHub page.

This would make a great ```dietpi-software``` addition. Is this something you'd like to try adding, or, want me to take a look? :)

R1CH-T, 2017-07-15:
@Fourdee 

Thanks for the quick response. ;)
...and sorry for the slow reply.

> I'll write up the steps needed to add this today, or, I can make the changes here and send the commit?

Thanks for the walk-through: You might want to consider adding this to the `README.md`.

> Fourdee added this to the v155 milestone 2 days ago

Well, It looks like I've got until v155 comes out, to get this sorted. :-)

If I don't get enough time, I'd appreciate it if you would do the honours, but it shouldn't come to that; I'll have a proper look at this ASAP and then post my attempt. At the end of the day, you're gonna have to satisfied with it (and test it, to make sure it doesn't break anything else), before you add it to the main release.

    https://github.com/ondrej1024/crelay
> This would make a great dietpi-software addition. Is this something you'd like to try adding, or, want me to take a look? :)

I want to have a go at adding this to `dietpi-software`, under _Hardware Projects_ and it shouldn't take very long for me to do so, but if you think that it should go elsewhere, it's up to you.

Thanks again, Dan.

Fourdee, 2017-07-16:
@R1CH-T 

Hi Rich,

> Thanks for the walk-through: You might want to consider adding this to the README.md.

Thanks, great suggestion 👍 

> If I don't get enough time, I'd appreciate it if you would do the honours, but it shouldn't come to that; I'll have a proper look at this ASAP and then post my attempt. At the end of the day, you're gonna have to be satisfied with it (and test it, to make sure it doesn't break anything else), before you add it to the main release.

Yep no worries, regardless, good stuff for giving it a go, most people don't lol 😄 . If you get no joy with it, let me know and i'll implement what you have and run some tests.

> https://github.com/ondrej1024/crelay I want to have a go at adding this to dietpi-software, under Hardware Projects, I intend to create a separate issue for this as-and-when.

Perfect 👍

R1CH-T, 2017-07-16:
Thanks @Fourdee.

...processing...

R1CH-T, 2017-07-18:
Hi, Dan: I've just spent a couple of hours on this (since 8:30ish) and here is my take; based of my current understanding:

Straight from my notes from this morning:

```
https://github.com/Fourdee/DietPi/issues/1075#issuecomment-315530565

Please look in https://github.com/google/aiyprojects-raspbian/tree/master/scripts.

Step 1.

In /DietPi/dietpi/func/dietpi-set_hardware:

Under "# - Disable known soundcard dtoverlays":

Add sed -i '/dtoverlay=i-googlevoicehat-soundcard/d' "$FP_RPI_CONFIG"

Under "Soundcard_Main()":

	#googlevoicehat-soundcard
			googlevoicehat-soundcard)

				# - enable dtoverlay
				echo -e "\ndtoverlay=$INPUT_DEVICE_VALUE" >> "$FP_RPI_CONFIG"

				# - i2s-mmap
				sed -i '/dtoverlay=i2s-mmap=/c\dtoverlay=i2s-mmap=on' "$FP_RPI_CONFIG"
				
				# - i2s
				sed -i '/dtoverlay=i2s=/c\dtoverlay=i2s=on' "$FP_RPI_CONFIG"
				
;;

Under "# - Apply asound.conf":

This is where I'm starting to get way out of my depth:  The script voice-recognizer-raspi/scripts/install-alsa-config.sh simply backs up the originals and then copies voice-recognizer-raspi/scripts/asound.conf into /home and /etc. I think that the contents of voice-recognizer-raspi/scripts/asound.conf should be added to the end of the #googlevoicehat-soundcard entry. i.e:

	#googlevoicehat-soundcard
			googlevoicehat-soundcard)

				# - enable dtoverlay
				echo -e "\ndtoverlay=$INPUT_DEVICE_VALUE" >> "$FP_RPI_CONFIG"

				# - i2s-mmap
				sed -i '/dtoverlay=i2s-mmap=/c\dtoverlay=i2s-mmap=on' "$FP_RPI_CONFIG"
				
				# - i2s
				sed -i '/dtoverlay=i2s=/c\dtoverlay=i2s=on' "$FP_RPI_CONFIG"
				
				# - Apply asound.conf	

						cat << _EOF_ > /etc/asound.conf
				
					options snd_rpi_googlemihat_soundcard index=0 # - I Don't know if this applies...

					pcm.softvol {
					type softvol
					slave.pcm dmix
					control {
					name Master
					card 0
					}
				}

				pcm.micboost {
					type route
					slave.pcm dsnoop
					ttable {
						0.0 30.0
						1.1 30.0
					}
				}

				pcm.!default {
					type asym
					playback.pcm "plug:softvol"
					capture.pcm "plug:micboost"
				}

				ctl.!default {
					type hw
					card 0
				}
				_EOF_

Step 2.

In DietPi/dietpi/dietpi-config

Under "Menu_AudioOptions()":

Add the menu option:
whiptail_menu_array+=("googlevoicehat-soundcard" "Google Raspberry Pi Voice Hat Audio")
```

Please take a look and see if any of it is OK, based off of Google's own method; which seems to work.

Fourdee, 2017-07-18:
@R1CH-T 

Great work 👍 

> options snd_rpi_googlemihat_soundcard index=0 #

Should not be required. In DietPi, enabling an overlay will disable the onboard soundcard, resulting in the addon card being the 1st device index ```0```. 
I also believe this would need to be added to module run-time options (eg, create file with contents of above: ```/etc/modprobe.d/googlemihat_soundcard.conf```.

> The script voice-recognizer-raspi/scripts/install-alsa-config.sh simply backs up the originals and then copies voice-recognizer-raspi/scripts/asound.conf into /home and /etc.

Tricky. Quick solution may be be to remove that reference in the install script, prior to running it, eg:
```
sed -i '/asound.conf/d' installscript.sh # Will delete all lines with "asound.conf" in "installscript.sh"
```

> # - Apply asound.conf	

This needs to be put at the end of ```Soundcard_Main()```, as it will be overwritten currently:
```
		# - Apply asound.conf
		if [ "$INPUT_DEVICE_VALUE" = "googlevoicehat-soundcard" ]; then
		
			cat << _EOF_ > /etc/asound.conf
options snd_rpi_googlemihat_soundcard index=0 # - I Don't know if this applies...

pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card $SOUNDCARD_TARGET_CARD
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card $SOUNDCARD_TARGET_CARD
}
_EOF_

		elif (( $ALSA_EQ_ENABLED )); then

			cat << _EOF_ > /etc/asound.conf
ctl.eq {
    type equal
    controls "$HOME/.alsaequal.bin"
}

pcm.plugequal {
    type equal
    slave.pcm "plughw:$SOUNDCARD_TARGET_CARD,$SOUNDCARD_TARGET_DEVICE"
    controls "$HOME/.alsaequal.bin"
}

pcm.!default {
    type plug
    slave.pcm plugequal
    slave.channels 2
}

ctl.!default {
    type hw
    card $SOUNDCARD_TARGET_CARD
}
_EOF_

			if (( ! $(dpkg --get-selections | grep -ci -m1 '^libasound2-plugin-equal') )); then

				apt-get install -y libasound2-plugin-equal

			fi

		else

			cat << _EOF_ > /etc/asound.conf
pcm.!default {
    type hw
    card $SOUNDCARD_TARGET_CARD
    device $SOUNDCARD_TARGET_DEVICE
}
ctl.!default {
    type hw
    card $SOUNDCARD_TARGET_CARD
}
_EOF_

		fi

```

> # - i2s-mmap
				sed -i '/dtoverlay=i2s-mmap=/c\dtoverlay=i2s-mmap=on' "$FP_RPI_CONFIG"	
				# - i2s
				sed -i '/dtoverlay=i2s=/c\dtoverlay=i2s=on' "$FP_RPI_CONFIG"

Just noticed these are additional overlays, that wont exist in ```config.txt```. So we'll need to add entries for these instead:
```
echo -e "dtoverlay=i2s-mmap=on" >> "$FP_RPI_CONFIG"
echo -e "dtoverlay=i2s=on" >> "$FP_RPI_CONFIG"
```
Then also add removal lines in the reset section:
```
sed -i '/dtoverlay=i2s=/d' "$FP_RPI_CONFIG" # Will cover both items
```

R1CH-T, 2017-07-18:
This thread is starting to get _very long_ now, so I'll be as brief as I can be. :)

I've read over your last reply and I think I see what you're doing there and follow it ~99%.

I think It's about time you took charge of this one now, if that's OK. I now see that, with the possible exception of `scripts/install-deps.sh`,we should pretty much do away with the installation scripts entirely and handle all of this through DietPi's methods, including setting up the system services required (using the contents of `scripts/install-services.sh` as a guide). For example; the "status-led" services will not be required if the Voice Hat is not being used.

I know that you'd probably prefer to be able to test this on your own Voice Hat hardware, but I'm more than happy to do this. This will take forever if I try to write it all myself, in the limited time I have. From what I can see, we've got enough of this sorted out, that you might as well go ahead and give me something to test.

If I get positive results, both with and without the Voice Hat, we can try it out on a variety of hardware.

As the scope of this has now changed in my mind, I think that it might be best to call it:
```
aSOFTWARE_WHIP_NAME[$index_current]='Google Assistant'
aSOFTWARE_WHIP_DESC[$index_current]='Google Assistant, including the AIY Kit for Raspberry Pi.'
```
And maybe miss out the `# - Disabled for All non-rpi` for the time being.

Let me know what you think.

R1CH-T, 2017-07-21:
Hi, Dan.

I got to the stage, three days ago where I started to think "This is way to complicated for me." and kinda bailed. It was unfair of me to be throwing this thing into your lap, especially as you have your own busy schedule and other considerations and priorities (as well as a roadmap to follow).

I basically realised that there were, very likely, several more pieces to this jigsaw that I hadn't even considered and I needed to take a break.

After taking this break and giving it a couple of days, I realised that the best way for me to get my head around this, would be for me to have a stab at working at an installer for the [**Crelay** driver / controller / web interface](http://github.com/ondrej1024/crelay); [which I mentioned before](https://github.com/Fourdee/DietPi/issues/1065#issuecomment-315336522).

To this end, I have opened an issue for this at https://github.com/Fourdee/DietPi/issues/1091.

We have covered the installation stage for the software and drivers here, but the parts of the process I hadn't considered, up until now, are _updating and uninstalling_. (I think that you might be able to give me some pointers around this in the other thread).

I also need to test the stuff we have discussed here, too. To this end, I have set up a DietPi image on a Pi3 running [BerryBoot](http://berryboot.alexgoldcheidt.com/images), which I have configured, updated and then cloned, in order to make repeated reinstallations fast and manageable, directly on a RasPi. (Thanks for that work, by the way). I will try to devote as much time as I can to these, this weekend, as I'm going for an "Unplugged Holiday" soon, so I need to get my affairs in order :). As you said: It can wait until "version-whatever", but I would like to include this soon.

Please take a look at the links above and in a couple of days, give me some feedback and I will hopefully be able to use any insights that I can gain from the process, to help me figure out all of the steps that I need to follow here.

Fourdee, 2017-07-21:
@R1CH-T 

> If you are on a deadline, please disregard this.

We don't do deadlines here "its ready when its ready". RL comes first :)

> I got to the stage, three days ago where I started to think "This is way to complicated for me"

Hehe, you probably picked the hardest installation to do, I'll probably struggle with this one lol.

> Please take a look at the links above and over the next couple of days, give me some feedback so that I will hopefully be able to use any insights that I can gain from the process, and help me figure out all of the steps that I need to follow here.

No worries. Once I get v154 out the door, I'll pick up this ticket and see if I can find a way to get this added into DietPi.

mpember, 2017-08-08:
I realise that my questions is not related to the SDK.  However...  does DietPi support the hardware from the MagPi kit?  Is there any specific steps required to get the sound / mic working?

I ask because I am looking to switch away from the 'official' image and build something that is a hybrid music player / assistant.  I've been using DietPi as the base for every other Pi-based project.  Before I ditch the AIY setup, I wanted to know if it was possible to use DietPi as the base for my build.

R1CH-T, 2017-08-17:
Hey, @mpember,

Sorry for the delay; I was away all last week and have just gotten the chance to reply.

> However... does DietPi support the hardware from the MagPi kit? Is there any specific steps required to get the sound / mic working?

My intention is for this option to be usable for both the AIY kit and other sound hardware, too. This means adding options for the hardware to DietPi's menus in order to facilitate this.

The steps I laid out earlier definitely work to get the AIY kit working perfectly in DietPi, but I'm just waiting for @Fourdee to get back on board, before we can proceed any further. ([See Here](https://github.com/Fourdee/DietPi/labels/ARMbaby).)

> After my initial post, I realised that it may appear like a hit-n-run from someone just looking for a shortcut.

Not at all. As you can see from the link above ^, Dan has a new project on the go and needs to devote all of his time to this at the moment. ;)

He is usually very prompt in responding to issues and requests. :)

> The main reason for my asking about separating the hardware from the software aspect of the Google stuff is that there are a wide range of AI programs out there. While I am not in any way asking DietPi to add all the non-Google programs, it would be great to simply have the hardware aspect of the install dealt with in the 'normal' DietPi way and not have to worry about manually cleaning up the Google AIY software stuff. The previous comments suggest that it is the Google software / config files that are the main cause of the remaining coding hurdles.

The driver installation, etc. is carried out by a few short scripts and config files, whose functionality needs to be broken out and handled by DietPi, thus enabling selection of various sound hardware to be (theoretically) possible.

> I have a few spare kits lying around and am willing to build one as a test. Unless I break it in my attempts to add stereo sound to it, I'm more than willing to use one as a test platform and offer any coding assistance where possible. I've got some basic "code monkey see, code monkey do" skills. If there is anything I can do to help out, just point me in the right direction and I'll start getting my hands dirty.

Sounds a lot like me!

Your input would be exceptionally welcome!

I certainly bit off more than I could chew by having a stab at this, but it has been a great learning experience, all the same!

Cheers,
Rich.

mpember, 2017-08-17:
@R1CH-T 

Thanks for the reply.

I saw the post about the latest @Fourdee 'project'.  It is a welcome reminder that even developers can have a real life.  👍 

For the moment, I've stuck with using the Google Assistant software.  By coincidence, your post appeared in my inbox just as I finished adding the Kodi-related voice commands to my AIY setup.

The (rather one-sided) discussion of what I've done with the kit can be found here:
https://forums.whirlpool.net.au/forum-replies.cfm?t=2653202

Since I have found out how to have the LED stop strobing when inactive, the drive to ditch the Google Assistant has lessened.  I would still like to see if I can replicate the functionality I now have on one of the third-party AI systems.

Once I get the core functionality working in the 'official' distro, I will set about building a test build.  With a 5 y/o niece, the plan is to give her something that is a bit less Google-centric.  I would rather not have her growing up starting every sentence with 'OK Google'. 😄 

In the meantime, the actions.py code I've been using is available here:
https://cloud.pembo.it/index.php/s/DfwT8ng9V3Pmxqr

R1CH-T, 2017-08-22:
Thanks @mpember. :+1: 

Sorry for the delay.

I really like the idea of the Kodi stuff and will take a look at your links ASAP. (Probably at the weekend.)

I have deliberately avoided following your links for the last few days, because I just haven't had the time to do so and I am avoiding Rabbit-Holes. (I hope you understand.)

Having been peripherally involved in the Kodi project for a number of years; ie. having donated, contributed and used the software regularly, I am really interested in voice-bindings for it.

I need to read your posts, in order to fully understand the context of your comment,above, but will try to do so soon.

Thanks,
Rich.

mpember, 2017-08-22:
(Apologies for the non exactly DietPi-specific reply)

Even though I have spent the last couple of weeks fleshing out the Google Assistant side of things, I am now seriously considering moving to one of the other AI projects.  The Google SDK limits you to a single trigger method.  This means that switching from the GPIO method to the use of a 'hotword ("OK Google") results in the inability to performs tasks when the button is pressed.

For most things, this isn't a major issues.  E.g. I have been able to write my own functionality for having audio podcasts and radio streams stop playing.  Since these tasks are handled using VLC for audio playback, it isn't too hard to just monitor the process and have a listener wait for a button press.  Google even have built-in functionality for adding an event listener.  But if I want to use MPD or Airplay, the limitation becomes a bigger hurdle.  It would be handy to use the button to pause playback and then wait for a voice command, maybe even resuming playback if no voice command executes within a certain time.  This requires modifying the core SDK code to have it listen for a GPIO trigger AND hotword trigger.  And modification to the SDK code is lost after any update to the SDK.

Since I have used the headers on the voice hat to add a second dac/amp/speaker, music playback is one of the main roles the device will perform.

Once I finish putting the Google/raspbian-based kit into the custom case, I will rebuild the spare kit and set about testing Diet-Pi as the base for the use of third-party AI software.

Thanks

mpember, 2017-09-09:
I have got it running.  It took a bit longer than I had hoped.  But that was mostly due to my unfamiliarity with the DietPi scripts.  I checked my modified version of dietpi-software a zillion times before I found the hardcoded limit on the number of software entries.

Firstly, the script fails at the "check_connection" test.  I think this is because the https://www.googleapis.com/auth/assistant-sdk-prototype URL returns a 404 response to any GET requests.

The only remaining issue with the system is that the voice-recogniser service needs to be restarted at boot.  The logs say the service has started, but the LED strobes like it is still booting.  After a restart of the service, the LED does the slow strobe and the service responds to voice prompts as usual.

mpember, 2017-09-28:
Just an FYI...

Google seem to have reworked their code and created a new branch with a new API.

The good news is that they appear to have rewritten their install scripts to allow for non-standard install paths.

I haven't had a chance to give the new code a go.  I will have a try over the weekend.  They say there is no migration path from the old to the new code layout / API, so you are required to do a clean install.

R1CH-T, 2017-10-14:
Thanks for the heads-up @mpember.

I do actually remember hearing something about this a while back, but didn't have the chance to check it out. I'll have to look it over.

Did you manage to try it out, in the end?

mpember, 2017-10-14:
I did.  Since it was a bit of a hatchet job (by me), I can't be 100% sure that the steps are 100% accurate.  That said...

_Remove the old service configs.  The new codebase is not designed to use the fragmented service structure._
`sudo systemctl disable status-led.service`
`sudo systemctl disable status-led-on.service`
`sudo systemctl disable status-led-off.service`
`sudo systemctl disable status-monitor.service`
`sudo systemctl disable voice-recognizer.service`

_Download the new AIY code_
`git clone -b voicekit --single-branch https://github.com/google/aiyprojects-raspbian.git  /mnt/dietpi_userdata/voicekit`

_Install driver and ALSA config (If doing a clean install)_
`cd /mnt/dietpi_userdata/voicekit`
`sudo scripts/configure-driver.sh`
`sudo scripts/install-alsa-config.sh`

There is only one file in the new "voicekit" code that still uses a hard-coded home path.  I have raised this as an issue with Google and gotten no response.
https://github.com/google/aiyprojects-raspbian/issues/139

The new code only comes with examples for a service script using the new structure.  They no longer provide a systemd config file  Their assumption is that developers should be integrating the code into a custom project and not simply cloning a Google-provided service.  If you want to see the files that I use, they are [available here](https://cloud.pembo.it/index.php/s/gFzsZm2ueFfyc7H).

The bit I didn't document was the rebuilding of the 'env' contents for the new code.  I butchered it due to my own incompetence and it is likely someone with a better knowledge of python will know exactly which steps are required.

I have successfully upgraded both of my AIY kits to use the new system and they are working fine.

If there are any questions, just shout.

Fourdee, 2017-10-14:
@R1CH-T 

Finally got one on order. Once it arrives i'll be able to get involved in this ticket.
https://twitter.com/DietPi_/status/914686122948689921

R1CH-T, 2017-10-26:
@Fourdee 
Cool.
Sorry, I missed this update. :(
I look forward to hearing your thoughts.

mpember, 2017-10-29:
It looks like the voicekit has been updated to contain a systemd service config file and the expected script name for the service will remain as **main.py** (in the **/mnt/dietpi_data/voice-recognizer-raspi/src** folder.

Since the instructions are to copy one of the examples to main.py, it may be worth chucking a note somewhere to inform users where to locate the files.  I understand that most DietPi users with a voicekit are likely to familiar with the setup, but it should protect against the odd 'I installed it and it doesn't work' forum posts.

Fourdee, 2017-11-11:
### Notes:
- online doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=9486#p9486

Fourdee, 2017-11-11:
Lightweight justice 😄 
[image]

Fourdee, 2017-11-11:
Pretty much done. 

Will run one more test install to check installation and online guide.

Fourdee, 2017-11-11:
@R1CH-T

I now see why the Google AIY kit is worth a purchase, so much fun 👍

mpember, 2017-11-11:
Is there a need to have the service run as **root**?  Mine is working fine as the **dietpi** user.

Fourdee, 2017-11-12:
Tested and completed.

R1CH-T, 2017-11-14:
Thank you so much for this, @Fourdee!

Also; thanks for your input, too, @mpember. It would not have been possible without you.

This stuff is going to be a moving target and is one of Google's top priorities, so I am in awe of what has been accomplished here.

> I now see why the Google AIY kit is worth a purchase, so much fun :+1:

I got lucky, in a way: When I subscribed to the MAGPI and got this kit, a year ago, I immediately jumped on the hype train. As the magazine transitioned to what it is now and became a fully legitimate, commercial publication, I was happy to join in.

I really don't think that most people understand how much joy can be derived from these sorts of projects. It was so much fun playing with this.... :)

Ordinary users, like myself, can make small, but significant contributions to the development of these embedded platforms; but it takes lots of work and it is not easy at all.

Much Respect,

Rich

mpember, 2017-12-28:
FYI, Google have changed the default branch to be the new vision-based kit.  This means that the script needs to explicitly request the **voicekit** branch.

Fourdee, 2017-12-29:
@mpember 

Many thanks, confirmed with their online doc:
https://aiyprojects.withgoogle.com/voice/#makers-guide-1-1--source-code
```
Alternately, the project source is available on GitHub: https://github.com/google/aiyprojects-raspbian/tree/voicekit. Note it is released under the "voicekit" branch.
```

Fourdee, 2017-12-29:
Completed.

closed 2017-12-29 by Fourdee