MichaIng/DietPi #1066: dietpi-update 1 kills openvpn needlessly

xanview, 2017-07-10:
#### Required Information:
- DietPi Version: 151
- SBC Device: RPi 3 Model B
- Power supply used: iPad charger 
- SD card used: Samsung Evo 64GB
- Distro (EG: Jessie) | Stretch

#### Additional Information (if applicable):
- Software title? openvpn
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Expected behaviour:
When running dietpi-update 1 it should update the software without affecting existing openvpn running process(es).

#### Actual behaviour:
The Raspberry Pi disconnected from the VPN (waited 10 minutes, no reconnection). SSHing over local IP and checking systemctl status shows:

```
root@DietPi:~# systemctl status openvpn@example.com
● openvpn@example.com.service - OpenVPN connection to example.com
   Loaded: loaded (/lib/systemd/system/openvpn@.service; disabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/openvpn@.service.d
           └─override.conf
   Active: inactive (dead) since Mon 2017-07-10 14:56:52 UTC; 10min ago
     Docs: man:openvpn(8)
           https://community.openvpn.net/openvpn/wiki/Openvpn23ManPage
           https://community.openvpn.net/openvpn/wiki/HOWTO
 Main PID: 785 (code=exited, status=0/SUCCESS)
```

#### Steps to reproduce:
Create a new configuration /etc/openvpn/example.conf.conf, reboot to connect to the VPN and run dietpi-update 1 which will kill the VPN.

Fourdee, 2017-07-14:
@xanview 

> When running dietpi-update 1 it should update the software without affecting existing openvpn running process(es).

Thanks for the report 👍 

DietPi-Update will stop the OpenVPN server service, however, this should not effect any OpenVPN clients.

Can you confirm this is a OpenVPN client connection on the device?

xanview, 2017-07-14:
It is the client connection on the device which is being affected :( - I have a config file /etc/openvpn/testdomain.conf which starts up at boot. When I run dietpi-update, this client connection is killed.

k-plan, 2017-07-23:
> Expected behaviour:
When running dietpi-update 1 it should update the software without affecting existing openvpn running process(es).

> Actualbehaviour:
The Raspberry Pi disconnected from the VPN during dietpi-update 1, I waited 10 minutes, no reconnection :( - SSHing over local IP and checking systemctl status shows:

Maybe this will help you out: 

[Cannot update through OpenVPN](http://dietpi.com/phpbb/viewtopic.php?f=12&t=960&p=5106&hilit=update+openvpn#p4150)

Can you please try:

`dietpi-update 2`

`dietpi-update 1 && reboot`

Fourdee, 2018-02-10:
Marking as closed, in favor of: https://github.com/Fourdee/DietPi/issues/1501

closed 2018-02-10 by Fourdee