MichaIng/DietPi #1068: [Question] Dependence on package "pciutils"?

MichaIng, 2017-07-13:
Is dietpi actually (still) dependent on the package "pciutils" respectively it's provided commands?

I do not find any use of `lspci`, `setpci` or `update-pciids` inside the code, nor any other use of the package: https://github.com/pciutils/pciutils

On my Pi2 I tried to run `lspci` with this result:
> pcilib: Cannot open /proc/bus/pci
> lspci: Cannot find any working access method.

I mean it doesn't disturb aether, but in case of looking for diet perfection... ;).

Fourdee, 2017-07-13:
@MichaIng 

> I do not find any use of lspci, setpci or update-pciids inside the code, nor any other use of the package: https://github.com/pciutils/pciutils

Great find, its not needed 👍 

Can be removed from system prep script and we can also patch for its removal.

Fourdee, 2017-07-14:
@MichaIng 

Resolved with your commit 👍 : https://github.com/Fourdee/DietPi/issues/1070

Will mark this as completed.

closed 2017-07-14 by Fourdee