MichaIng/DietPi #1069: XRDP install doesn't work unless TightVNC is installed

FutureTense, 2017-07-13:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```153```
- SBC Device ODROID C1+
- Power supply: USB
- SD card used EEMC
- Distro (EG: Jessie) | ```Linux DietPi 3.10.104 #1 SMP PREEMPT Sun Mar 26 14:55:18 CEST 2017 armv7l GNU/Linux```

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

Should be able to launch desktop using Windows Remote Desktop
#### Actual behaviour:

See description and screenshot posted on [forum](http://dietpi.com/phpbb/viewtopic.php?f=11&t=2053)

#### Steps to reproduce:

Fresh DietPi install.
Install a desktop, such as MATE
Install XRDP
Start a remote desktop connection from Windows

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

From a bit of research I discovered others fixed this by installing TightVNC before installing XRDP.  I uninstalled XRDP, installed TightVNC then installed XRDP and it worked.  I did not try installing TightVNC after installing XRDP

Fourdee, 2017-07-13:
@FutureTense 

Thanks 👍 

> https://github.com/Fourdee/DietPi/issues/1071

Fourdee, 2017-07-14:
Notes:
- Straight forward apt install ```xrdp```
- Possible apt installation issue, would explain the install of tightvnc to resolve?

Tests:
- Fresh DietPi image, install XRDP only
- Fresh DietPi image, install MATE & XRDP.

Fourdee, 2017-07-14:
@FutureTense 

We were unable to replicate the issue in our tests. We believe this may be limited to an APT installation issue, or a unknown connection issue with XRDP server (in which case we would need logs to debug).

- Can you remember seeing any onscreen apt errors during the 1st installation (eg: prompt), when the connection failed?
- If the connection issue occurs again, please can you paste the results of the XRDP logs:
```cat /var/log/xrdp-sesman.log```

FutureTense, 2017-07-17:
I'll try a fresh install when I get a chance. Please leave this open for a few weeks?

Fourdee, 2017-07-17:
@FutureTense 

> I'll try a fresh install when I get a chance. Please leave this open for a few weeks?

Yep no worries, please let us know results 👍

MichaIng, 2018-01-31:
@FutureTense
Is this issue resolved for you? Otherwise now is a good time for a fresh installation and retest, as v6.0 was released which can't be updated to from 159 or lower.

Fourdee, 2018-01-31:
@MichaIng 

May be linked to similar issue with https://github.com/Fourdee/DietPi/issues/1442

However, i'll close this in preference of the above, which I'll test/verify/resolve installation.

closed 2018-01-31 by Fourdee