MichaIng/DietPi #1072: XU4] Kernel 4.9 Unstable Cloud Shell HDD

studioai, 2017-07-14:
- DietPi Version  v153
- SBC Device  XU4
- Power supply used  5v6a
- Distro (EG: Jessie) | ```uname -a```? Jessie (4.9 Test Kernel)

I'm testing new 4.9 kernel. and I met an issue with cloudshell hdd.

https://forum.odroid.com/viewtopic.php?f=146&t=27548

my issue exactly same with forum thread. (Even PSU is same)

I've never experienced this issue when I was using 3.10 Kernel.

Fourdee, 2017-07-15:
@studioai 

Thanks for the report and info. Seems a known issue with 4.9 kernel and Odroid are aware, have you tried any of the workarounds in https://forum.odroid.com/viewtopic.php?f=146&t=27548 ?

```
Failed to process /etc/kernel/postinst.d at /var/lib/dpkg/info/linux-image-3.10.105+.postinst line 391.
debconf: DbDriver "templatedb": could not write /var/cache/debconf/templates.dat-new: Read-only file system
```
Indicates rootFS is mounted as read only, possibly due to an issue with the file system. Please can you paste results:
```
cat /etc/fstab
mount
df
dmesg | grep mmc
```

studioai, 2017-07-15:
Thnaks. I'll Try to downgrade.

studioai, 2017-07-15:
Here's results

```
root@DietPi:~# cat /etc/fstab
#Internal Drives---------------------------------------------------
proc            /proc           proc    defaults											0 0
UUID=5886-8548  /boot           auto    defaults,noatime,discard							0 2
UUID=a4f12232-1670-43f4-8eca-49bfbe4d01ee  /               auto    defaults,noatime,discard							0 1
tmpfs 			/tmp  			tmpfs 	defaults,noatime,nodev,nosuid,mode=1777				0 0
tmpfs 			/var/log 		tmpfs 	defaults,size=20m,noatime,nodev,nosuid,mode=1777	0 0
tmpfs 			/DietPi 		tmpfs 	defaults,size=10m,noatime,nodev,nosuid,mode=1777	0 0

#External Drives---------------------------------------------------
#NB: Please use dietpi-drive_manager to setup and control your external drives.

#Samba Client------------------------------------------------------
#/mnt/samba . Please use dietpi-config and the Networking Options: NAS menu to setup this mount

#FTP Client Mount--------------------------------------------------
#/mnt/ftp_client . Please use dietpi-config and the Networking Options: NAS menu to setup this mount

#NFS Client Mount--------------------------------------------------
#/mnt/nfs_client . Please use dietpi-config and the Networking Options: NAS menu to setup this mount
UUID=523139b0-c49e-42ab-a3c1-38d2243e752b       /mnt/usb_1      auto     defaults,noatime,nofail,x-systemd.automount  0 0
UUID=e4e4c294-65f6-498e-b980-2aa7d5759130       /mnt/usb_2      auto     defaults,noatime,nofail,x-systemd.automount  0 0

# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
root@DietPi:~# mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
udev on /dev type devtmpfs (rw,relatime,size=10240k,nr_inodes=187661,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,relatime,size=408792k,mode=755)
/dev/mmcblk1p2 on / type ext4 (rw,noatime,discard,data=ordered)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=23,pgrp=1,timeout=300,minproto=5,maxproto=5,direct)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
mqueue on /dev/mqueue type mqueue (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
tmpfs on /DietPi type tmpfs (rw,nosuid,nodev,noatime,size=10240k)
tmpfs on /var/log type tmpfs (rw,nosuid,nodev,noatime,size=20480k)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/mmcblk1p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,utf8,errors=remount-ro,discard)
/dev/sda on /mnt/usb_1 type ext4 (rw,noatime,data=ordered)
/dev/sdb on /mnt/usb_2 type ext4 (rw,noatime,data=ordered)
tmpfs on /run/user/995 type tmpfs (rw,nosuid,nodev,relatime,size=204396k,mode=700,uid=995,gid=995)
tmpfs on /run/user/0 type tmpfs (rw,nosuid,nodev,relatime,size=204396k,mode=700)
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,relatime)
root@DietPi:~# df
Filesystem      1K-blocks      Used  Available Use% Mounted on
udev                10240         0      10240   0% /dev
tmpfs              408792      7148     401644   2% /run
/dev/mmcblk1p2   30280756   3751316   25589100  13% /
tmpfs             1021976         8    1021968   1% /dev/shm
tmpfs                5120         4       5116   1% /run/lock
tmpfs             1021976         0    1021976   0% /sys/fs/cgroup
tmpfs               10240      1364       8876  14% /DietPi
tmpfs               20480      1468      19012   8% /var/log
tmpfs             1021976         8    1021968   1% /tmp
/dev/mmcblk1p1      72542     33174      39368  46% /boot
/dev/sda        961303576 546814632  414472560  57% /mnt/usb_1
/dev/sdb       1922729860 806867968 1115845508  42% /mnt/usb_2
tmpfs              204396         0     204396   0% /run/user/995
tmpfs              204396         0     204396   0% /run/user/0

```

Fourdee, 2017-07-17:
@studioai 

Thanks 👍 

Everything looks fine, no mention of read only. Can you try the kernel downgrade again?

studioai, 2017-07-17:
@Fourdee 

No Error occurred but failed to boot. 

I cannot report what is the problem because nothing on my display. Sorry.

Fourdee, 2017-07-17:
@studioai 

> No Error occurred but failed to boot.
I cannot report what is the problem because nothing on my display. Sorry.

Considering the issues reported with read only filesystem, I can only assume a possible filesystem corruption due to unstable system. Surprising as your running the official 5V/6A PSU? We tend to see this with 5V/4A with cloudshell (insufficient).

Are you using official EMMC, or SD card?

I'd recommend trying a fresh image write of DietPi.

studioai, 2017-07-18:
I'm using non-official 5v/6a PSU. (They didn't offer a PSU when I purchased C-shell).
Using a SD card.

I've already lost my git repositories cause I missed a backup file. and that's enough for me. I'll give up and wait for hard kernel's solution.

closed 2017-08-03 by studioai