MichaIng/DietPi #1073: Plex media server - will not install

bwaddington, 2017-07-15:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2017-07-15:
@bwaddington 

Hi, thanks for report.

We'll need more information.

- Which device are you running (eg: RPI 3)?
- Did you receive any on screen error prompts during the installation?

bwaddington, 2017-07-15:
Thank you for getting back to me! i have managed to sort it out, there was a problem with the mirror site. plex installed but there was no web interface. just did a clean install of dietpi and its all working now.

Fourdee, 2017-07-15:
@bwaddington 

Thanks for letting us know 👍 . I'll mark this as resolved.

closed 2017-07-15 by Fourdee