MichaIng/DietPi #1074: [Feature Request] Auto Reconnect Wifi in DietPi - Wireless Network Options

bhaveshgohel, 2017-07-15:
Dear @Fourdee ,
Is it possible to add Auto Reconnect Wifi in 'DietPi - Wireless Network Options' by setting Wireless Details in 'Manually Set Wireless Details' option. 

**Goal** : Raspberry Pi will keep on searching for the SSID at some interval and if the network available it will connect to that network and  when on disconnect it again search for the same network and repeat.

Fourdee, 2017-07-15:
@bhaveshgohel 

> Is it possible to add Auto Reconnect Wifi in 'DietPi - Wireless Network Options' by setting Wireless Details in 'Manually Set Wireless Details' option.

Currently not, if the SSID drops connection, WiFi will unlikely reconnect.

> Goal : Raspberry Pi will keep on searching for the SSID at some interval and if the network available it will connect to that network and when on disconnect it again search for the same network and repeat.

I made a script to monitor existing WiFi connections, and, reconnect if it fails. We could implement that system into DietPi-Config?
https://community.roonlabs.com/t/wifi-disappears-after-a-day/25850/20?u=dan_knight

bhaveshgohel, 2017-07-15:
Dear @Fourdee, 
I tested the script and it is working fine... :+1:  ( Thank you for adding in V155.  :tada: )

Fourdee, 2017-08-18:
Completed.

closed 2017-08-18 by Fourdee