MichaIng/DietPi #1076: XU4 - Unstable LAN since last update

hydrouk, 2017-07-15:
- DietPi Version - 153
- SBC Device XU4 + CloudShell
- Power supply used 5v 4A 
- Using eMMC 16GB
- Distro Linux NAS 3.10.104+ #1 SMP PREEMPT Wed Oct 26 16:17:46 CEST 2016 armv7l GNU/Linux

I updated to 153 two days ago and thought everything was running fine but discovered today that i couldn't connect to the sshd, i then noticed there was no LAN light anymore.  I just power cycled the machine and it came back up and was working again, one hour later i go to login again to check and the interface was down again.  I checked the logs and saw the following

Jul 15 11:17:19 NAS netdata: Cannot open file '/etc/netdata/netdata.conf'
Jul 15 11:17:34 NAS logger: /etc/dhcp/dhclient-exit-hooks returned non-zero exit status 2
Jul 15 18:17:15 NAS netdata: Cannot open file '/etc/netdata/netdata.conf'
Jul 15 19:43:32 NAS sshd[10171]: fatal: Read from socket failed: Connection reset by peer [preauth]

hydrouk, 2017-07-16:
Stayed alive over night but has been downloading also.

Fourdee, 2017-07-16:
@hydrouk 

> i then noticed there was no LAN light anymore

Must be limited to your system, either a local network issue (router/cable) or a hardware related issue on the XU4.

I'am yet to experience this with the XU4 on both 3.x and 4.9 kernels:
[image]

We'll need to get more info if we are debug. When the loss of connect occurs, we need to check basic networking status and dmesg for any errors, at the physical machine:
```
dmesg | grep eth
ifconfig -a

#Lets also test a ping
ping google.com

#Try bringing the adapter up again
ifdown eth0
ifup eth0
```

hydrouk, 2017-07-17:
It has been stable ever since last reboot which was 2 days ago, weird moment! I do appreciate you getting back to me though, thanks

closed 2017-07-17 by hydrouk